whaleygeek/pyenergenie #2: Need a way to persist the device database to disk

whaleygeek, 2015-10-03:
e.g. on each ADD, check a dbm file, and if device does not exist, add it.
Also need a way to list the contents of this database.

whaleygeek, 2016-03-17:
This will become more useful when I start testing the control-only MiHome adaptors, as there is no way to discover their address. It will be useful to be able to pre-seed the device directory with known device addresses so that these devices can also be made to work in the switch.py tester.

whaleygeek, 2016-03-27:
Noted that #31 has a good idea in it to basically build a DNS (name system) for devices and persist this - i.e. so you could refer to devices by friendly names such as 'microwave' rather than their device address. I think this is a great idea, and will consider it when I design the device database.

PatchworkBoy, 2016-03-28:
Might be worth looking at how HomeKit devices are handled within hap-nodejs repo...?

whaleygeek, 2016-04-15:
I plan to move the existing device directory from the monitor.py example into a separate module first, define this as a simple abstraction, and then let others choose which particular implementation they want to sit under this abstraction. I will provide a simple out-of-box file for persisting device details, but you can then always change this to use an SQL database or any other directory system of your choice.

whaleygeek, 2016-04-20:
It seems like a simple sqlite3 implementation might be useful here, as a version of sqlite3 is standard in the python distribution from 2.5 and above

whaleygeek, 2016-05-15:
In the unified_radio branch, which I will soon merge back to master, I have just moved this code into a energenie/Registry.py - this hives it off into a place independent of the application, and anyone could change this for their own implementation if they wish.

I'll close this issue when the unified_radio branch is merged back to master.

whaleygeek, 2016-05-16:
unified_radio merged to master.
If you want to persist the device database to disk, modify the energenie/Registry.py to add your own persistence functionality.

closed 2016-05-16 by whaleygeek