whaleygeek/pyenergenie #8: Investigate radio reset lockup

whaleygeek, 2015-10-03:
Occasionally the radio gets into a state that it stops working. Investigate availability of hardware reset line on GPIO, and reset the radio when the driver first starts.

cafebabe, 2015-10-06:
I've written this C code to perform a reset as described in section 7.2.2. of this document http://www.hoperf.com/upload/rf/RFM69W-V1.3.pdf.
```
#define RESET	RPI_V2_GPIO_P1_22

bcm2835_gpio_write(RESET, HIGH);
bcm2835_delayMicroseconds(100);
bcm2835_gpio_write(RESET, LOW);
bcm2835_delay(5);
```
Hope it helps.

whaleygeek, 2015-10-06:
Thanks. The schematic I got from Energenie didn't show the RESET line as connected, so I never really investigated this option. I'll ask them for an updated schematic as it might be that they connected it in a later revision of the board.

whaleygeek, 2015-10-07:
Yes, confirmed on latest schematic, this is GPIO25(pin22). I'll add in reset support so that the radio is always in a clean state at startup. I was having occasional lockup issues too, and I had to unplug and replace the radio to reset it (quicker than rebooting the Pi!)

whaleygeek, 2015-10-07:
I've just added a RESET on init. I've verified that this comes out on the right pin on the RPi. I've also seen a much more reliable operation of the radio now as a result of this, reports from the MiHome Adaptor Plus come out reliably every 10 seconds.

closed 2015-10-07 by whaleygeek