whaleygeek/pyenergenie #18: Modelling devices as classes

whaleygeek, 2015-10-14:
Initial Proposal for modelling devices as classes.

The purpose is to be able to write expressive and compact applications that talk in the land of physical devices such as 'myRadiator' and 'myPlug' or even second level names such as 'myKettle' (which is plugged into 'myPlug'). Thus hiding the detail of how messages get encoded and transported, and focusing an application more on it's intents than on it's implementation.

* each different device has a class 'cookie cutter' (e.g. Energenie.MiHomeAdaptorPlus(),Energenie.eTRV())

* each instance of a device is an object (e.g. plug, rad...)

* each device instance can be named and abstracted as appropriate to it's purpose (e.g. kitchenKettle(plug), hallwayRad(rad))

* Outgoing command messages modelled by function calls and/or properties (e.g. kitchenPlug.turnOn() or even kettle.turnOn()...)

* Incoming data modelled by function call getters and/or properties, perhaps with a cache of the last received value (e.g. hallRad.getTemperature(), hallRad.temperature)

* meta-model driven proxy classes for device types not in the device dictionary (e.g. an unknown device sends it's temperature, so you can still say unknownDevice.getTemperature() or unknownDevice.temperature)

* a low dependency link from the device object instances to the transport that sends and receives messages on their behalf (e.g. kitchenPlug = Energenie.MiHomeAdaptorPlus(OpenHEMS(myRadio))

* an outwards link via OpenHEMS(radio) for queueing transmit messages. An inwards link from OpenHEMS(radio) back to the object, so that when a message is received from a specific (mfrid, prodid, sensorid) the data is routed to the correct object instance.

* perhaps a catch-all object that receives messages from unknown or unregistered devices, with a way to later morph those into real device instances (e.g. this could be how the discovery service is created, by all messages for unknown sensorid's going to a catch all, and later the app constructing appropriate object abstractions around them and them being removed from 'unknown'

whaleygeek, 2016-04-15:
Also note that I intend this design to work directly with the (to be coded) message sequencer to ensure that messages are transmitted at 'quiet' times in the schedule to prevent collisions with reports from other devices. I also plan to completely abstract away in the device classes used by app whether the device is a MiHome (FSK) or Legacy (OOK) and automatically switch the radio modulation configuration and safely schedule outgoing messages.

This will allow building systems combined with both MiHome and Legacy devices, and treating them all as first class citizens in any user app.

I won't be starting work on this until I have finished the hardening of the physical layer.

whaleygeek, 2016-05-13:
I'm thinking of a device-agnostic interface, so you can just have python variables for any connected device (green button or purple), with a simple device registry. Then instigate methods on them. This will knit up to the radio interface under the bonnet for you, and switch between OOK and FSK mode automatically - mostly leaving the radio idling in FSK receive mode to increase the chances of hearing a report. Later I will add a message scheduler that learns the reporting schedule, and defers FSK or OOK transmits to 'free timeslots' to improve the performance of the overall system.

So something a bit like this maybe:

(when configuring the system)
print("Press the learn button on the TV device")
energenie.start_learn(house_code=0xABCDE, index=2)
raw_input("Press enter when done")
energenie.stop_learn()

energenie.create_device("tv", energenie.device.ENER002, index=2, house_code=0xABCDE)
energenie.create_device("aquarium", energenie.device.MIHO005, address=0x1234)

(when running the system)

tv = energenie.get("tv")
aquarium = energenie.get("aquarium")

tv.off()
aquarium.on()
time.sleep(10)
if aquarium.power > 20:
print("Has the pump motor stalled??")

def just_turned_off(device):
print("Your user just turned off %s" % device)
print("I'm turning it back on!")
device.on()

aquarium.when_turned_off(just_turned_off)

whaleygeek, 2016-05-13:
This is four separate pieces of work:

1) a persistent device registry
To allow device types and modulation (ENER, MIHO) and (FSK, OOK) plus addressing information, to be associated with a friendly name 'tv', and persisted to a disk version of a registry database. This database to be queryable so that runtime device classes can be reconstructed from their name in this database.

2) a device class for every supported device
To allow device specific intents to be mapped onto on air messages, and for incoming data to be cached locally in RAM to allow simple on demand query semantics.

3) device class to radio interface
To allow device classes to be controlled by intents such as tv.turn_on() and for that intent to trigger some on-air activity in the radio driver. Also, for received messages from the on-air interface to be routed to the device class that handles that device, so that when the MiHome adaptor plus address 0x123 reports it's energy, that energy report is cached inside the 'tv' object, allowing deferred queries like tv.is_on() and tv.get_power()

4) A dynamic message scheduler that intelligently switches the radio between transmit and receive and OOK and FSK modes. It will learn the reporting times of each device that is in earshot, build a timing plan, and use that timing plan to defer later transits to timeslots that are not already occupied by a regular report for a device, and thus decreasing the probability of missing a rx due to being in tx, and therefore improving the performance of the overall system.

whaleygeek, 2016-05-15:
The unified_radio branch was the blocker for this, but that work is completed now. Once it merges to master, I'll start a new feature branch to start investigating and experimenting with the device class interface, and the underlying connection to the physical layer radio interface.

whaleygeek, 2016-05-15:
One of the other things about this, is that the registry record for a device instance can relate it to a device class (in Devices.py). Classes in Devices.py could then define the radio parameters needed in order to communicate with that type of device (e.g. a purple mihome could be (FSK, repeats=1) and a green button could be (OOK, repeats=8) etc).

Then when the app tries to use tv.on() it really consults the parameters in it's parent class, a MIHO005, which configures the radio as FSK,repeats=1 and sends the message. When the app tries to use fan.on() it consults the parent class, gets an ENER002 which configures the radio as OOK,repeats=8 and sends the message.

These device classes could also estimate based on the message you are sending, how long it will take to send that message, and it can use that when (later) consulting an intelligent message scheduler to work out a timeslot that it can use that will be long enough to *not* collide with a known report time from another MiHome device that is known to be sending regular reports.

Also, the device instance (tv, fan) can then store 'last heard from time' to work out if the device is working or not, and also 'last values' such as to service any requests such as get_power() and get_voltage() and get_frequency() - i.e. these accessors will read the last known value (or raise an error if no value is known) rather than trying to request it on demand - many devices do not support request on demand, so they have to be cached. They can be cached with a receipt timestamp, so that the app can always put an expectation on how up to date the data is that it requires.

whaleygeek, 2016-05-17:
This work has started on the device_classes branch

whaleygeek, 2016-05-17:
The device_classes branch has the first cut proposed object oriented interface, with all the method signatures defined.

Files: classes.txt, Devices.py and Registry.py

This is not the final API, but it's the 'best first attempt' at defining the method signatures and requirements.

Note I have proposed a way to auto create class variables from the registry, to save the app writer from needing to do all that mucky initialisation stuff at the start.

I think it might work something like this:

1. stuff a load of devices in the registry, either by creating a config file, or by learning them with discovery, or by programmatically discovering or coding them.

2. run your program, using this line in the main app

from Registry import registry
registry.auto_create(modules[__file__])

3. Then just use the auto created variables in the context you passed to auto_create (in this case, the module)

tv.turn_on()

i.e. if you have a device in your registry called "tv" then auto_create will create you a variable called tv, create the class instance wrapper for the correct object, and connect the two together.

Details of how the registry gets access to the radio interface in order to auto connect device class instances to the radio driver is still TBD, but it will probably work on a 'sensible default' with a way to override it if you want to mock the radio out for auto testing purposes.

iotic-labs-markwharton, 2016-05-18:
This all looks great.  The only bit I'm unclear on is discovery?  Would I register a callback to be called when a new device is discovered and then call a factory method to get a new one?  That would be nice then I could use isinstanceof() to find out what kind of energenie it was.  Poorly thought out thought, I know.

whaleygeek, 2016-05-18:
Thanks @iotic-labs-markwharton - good thought. 

Discovery is something that I haven't completely planned out an API for yet, I was just bumping into the thinking myself that a discovery mechanism would be helpful, and roughly where it might knit into. I'll open a new issue about Discovery and will capture the thoughts there, as it will *probably* come in after the Device classes and registry are mostly implemented.

whaleygeek, 2016-05-18:
Discovery of devices logged as #56 as I think it will be a good feature to add once device classes and the registry are mostly working (and I think that some of the thinking will need to be done at this stage to make sure it will dovetail in nicely later)

whaleygeek, 2016-05-18:
The work on the device_classes branch is just about bursting into some life with a mocking interface.

Here's how the registry auto creation process has come out in the wash:

```
    # seed the registry
    registry.add(Devices.MIHO005(device_id=0x68b), "tv")
    registry.add(Devices.ENER002(device_id=(0xC8C8C, 1)), "fan")

    # test the auto create mechanism
    import sys
    registry.auto_create(sys.modules[__name__])

    # variables should now be created in module scope
    print(tv)
    print(fan)

    tv.turn_on()
    fan.turn_on()
```

whaleygeek, 2016-05-23:
Work on the device_classes branch is progressing well.

I have finished a new OpenThings.Message() abstraction that makes it much easier to edit and query messages, this passes all it's test cases, and it is integrated into the 4 testers legacy.py combined.py switch.py and monitor.py.

The Registry (non persisted) is pretty much working, as is the Devices.py that describes the facilities of the ENER002 and MIHO005 devices. These two device classes are correctly in the device hierarchy as a LegacyDevice and a MiHomeDevice respectively.

The AirInterface architecture is working, and the TwoBit and OpenThings air interfaces are correctly adapting message flows through the pipeline right out to the radio and back.

The 4 testers are updated to use all of this new architecture. This has just been tested on a Raspberry Pi and it all works as expected.

The work on this branch is not finished yet, but here is the githash of the tests just performed on the RaspberryPi if anyone else wanted to try out an early version of this work themselves.

https://github.com/whaleygeek/pyenergenie/tree/6f6df45f92cf8de68f497253cbb0ae59100c0862

whaleygeek, 2016-05-23:
Next steps are to use the ENER002 and MIHO005 device classes in all 4 tester applications, and make the applications completely object oriented. A lot of the message building and routing code will disappear from these 4 testers as a result and they will probably shrink quite a lot.

A decision has to be made as to how to handle receive concurrency. I suspect in the first tests of this, I will have a loop() that is called to put the radio into receive mode. Some simple algorithm will need to be used to put the radio into FSK receive on a known period (or perhaps at this point in the tests it is just done asyncrhonously). Feels like it should be left in receive all the while there is no transmit to do (and the radio can be left in receive and polled anyway in the existing architecture).

Work is required on the Registry.Router class to wire receive paths based on mfrid+productid+deviceid through to device class instances created from the registry expansion process, so that the correct instance can handle the messages.

The main loop will then poll the device classes (or use an on_updated() callback) to process received data as it arrives.

My current estimates say this might take 2-3 days to complete, so this work *might* be finished and ready for merge by Friday 27th May if all goes to plan, and this will make the full OO example applications (and the complete hiding of the radio message interface) possible. This will make the demo apps as simple to read (and write) as the GPIO based tx-only Energenie radio example code.

whaleygeek, 2016-05-24:
Work is progressing well on device classes.
The new Message() interface is working really well, as is the fsk_router, the registry, and the device classes.

I have re-coded all the examples to use the new, simpler, OO interface, and injected synthetic receive messages for testing, and captured output payloads for tx testing. It all seems to work.

There's a bit more to do, mostly centring around making the device registry persisted so that it can resume between runs of a program, and looking at device discovery.

There's also a bit of re-connecting to do for the MiHome join messages, and some optional workflows to add to enable or disable auto registration of newly discovered devices, and auto responding with join_ack to join requests (these are features I think the app should turn on or off as needed)

Still on track for merging this back roughy Friday 27th May

whaleygeek, 2016-05-27:
Most of the work on the device registry persistence is done, although I don't think this will release today as there is more testing to do.

Remaining work:
- finish test cases to test the registry auto load from file
- add in support for auto receive routing knit-up when restoring registry from file
- write a number of better application examples using the new registry and object classes
- re-test everything on real devices
- merge back to master

Given the bank holiday weekend in the way, it might be some time mid week w/c 30th May that this surfaces to master with all the apps written and all the device testing completed.

For those of you eager to make progress, you could just clone from the device_classes branch and put up without registry persistence for now, and start to get some mileage on your apps using the new OO classes.

whaleygeek, 2016-05-28:
The registry data store and the registry are now completely tested and working, including support for auto device rx routing for devices that transmit data, and auto variable creation in a context if required.

The remaining work is:
* writing better application examples
* testing on real hardware
* merge back to master

The API is pretty settled now, so those wishing to gain early access to this work could usefully clone the latest device_classes branch and start experimenting with that and building their own demo applications.

I'm going to be quite particular about testing with real devices and writing nice demo apps, so it might be another week before this merges back to master. But I doubt the API will change any more in that time.

I'm particularly pleased with the registry.load_into() method, which allows you to construct a registry via other means (e.g. by programmatically building it, or using the discovery mechanism in another mode or another program). You can then just do something like this in your actual app, to auto create device variables that you can interact with directly:

```
import sys
import energenie

energenie.init()

context = sys.modules[__name__]
energenie.registry.load_into(context)
xbox.turn_on()

while True:
    if xbox.get_power() > 25 and game_time_exceeded():
        xbox.turn_off()
        tv.turn_on()
        living_room_light.turn_on()
    energenie.loop()
```

whaleygeek, 2016-05-28:
The application examples are now specified:
```
setup_tool.py
mihome_energy_monitor.py
games_console_timer.py
web_console.py
```
I expect I will only write the setup_tool and also re-code the other standard examples before merging back to master.
```
control_legacy.py
control_mihome.py
control_both.py
monitor_mihome.py
discover_mihome.py
```

whaleygeek, 2016-05-29:
setup_tool.py is mostly finished and tested in simulation now.

I'll be looking through the existing demos (control_ monitor_ and discover_) to bring them in line with the new device class API, and using the setup_tool.py as a way to seed the registry with devices used by these applications (which will simplify them as examples).

I think full application examples might wait until this work has merged back to master, so that I can get the core functionality out there and people using it, and build some credible demos later once it has settled.

whaleygeek, 2016-05-29:
All the examples that I want written before mergeback are written now and tested as much as possible through simulation. 

This includes a new example 'control_any_auto.py' that shows how to use the auto variable creation mechanism of the registry - seed the registry with devices yourself by writing a registry.kvs by hand, or by running setup_tool.py to discover them. Then just run the program and variables named after the same name as the device in the registry will magically be created on script start, containing all the methods required to interact with that specific object. This keeps your application script very very small and simple.

The last thing to do before merge is to test on real hardware.

whaleygeek, 2016-05-29:
@iotic-labs-markwharton please review the discovery functionality described in #56 and in the code on the device_classes branch in setup_tool.py and discover_mihome.py, as I slotted this in with this work especially for you. If you could report back any observations in issue #56 that will be helpful, thanks.

whaleygeek, 2016-05-30:
The device_classes branch and sprint is now finished and merged back to master, and all seems to work good enough. See the various commit logs and other issue updates and the new README.md for instructions on what has changed and how to use it.

@iotic-labs-markwharton I think your discovery requirements are now dealt with, please see #56 for a description of how it all works.

closed 2016-05-30 by whaleygeek