whaleygeek/pyenergenie #20: Using the MIHO002 'Dumb' Switch

scousepi, 2016-01-08:
Hi, I'm trying to use some MIHO002 MiHome adapters http://tinyurl.com/zdyzud8 which are 'dumb' Energenie switches in as much as they don't broadcast back to a controller their switch state. The problem I have is that the only documentation I can find relates to using legacy OOK switching to access them meaning that (so it seems) only 4 switches can be addressed. You can however use more than 4 with the Mi|Home hub - suggesting a more complex form of addressing is possible. Would you have any suggestions about how I might find out some more about this, or at least send a broadcast message to see if the MIHO002 responds with any kind of device ID etc. when it is in learning mode. Sadly I know nothing about C or Python, but I am willing to learn and would love to get these switches working if I can.

LibanO, 2016-01-08:
Hi,
The gateway can broadcast more than 4 channels by changing the broadcast address which is the 20 bits prior to the 4 bit command. In this respect the number of sockets that can be control are no longer limited to 4. I can send you documentation and sample program if you wish. I've attached a guide and demo program
[TxDemoOOK.zip](https://github.com/whaleygeek/pyenergenie/files/82590/TxDemoOOK.zip)
[ENER002PI-RT User Guide updated.pdf](https://github.com/whaleygeek/pyenergenie/files/82591/ENER002PI-RT.User.Guide.updated.pdf)

Regards
Liban

scousepi, 2016-01-08:
That's fantastic. I will take a look at the demo program tonight. Thank-you very much!

scousepi, 2016-01-09:
Hi, I've been looking at this demo code and although I can get the precompiled txdemo to work, every time I run make to compile it locally I'm getting errors (below). I'm using a Raspberry Pi 2 (Quad Core) with the latest version of Raspbian Wheezy and bcm2835 v1.5 drivers (which do compile successfully). I've copied the errors below. Any advice would be appreciated...

Thanks

cc -Wall   -c -o main.o main.c
main.c: In function ‘main’:
main.c:33:20: error: ‘RPI_V2_GPIO_P1_13’ undeclared (first use in this function)
main.c:33:20: note: each undeclared identifier is reported only once for each function it appears in
main.c:34:20: error: ‘RPI_V2_GPIO_P1_15’ undeclared (first use in this function)
main.c: In function ‘HRF_send_OOK_msg’:
main.c:191:21: error: ‘RPI_V2_GPIO_P1_15’ undeclared (first use in this function)
main.c: In function ‘HRF_reg_Wn’:
main.c:230:2: warning: implicit declaration of function ‘bcm2835_spi_writenb’ [-Wimplicit-function-declaration]
<builtin>: recipe for target 'main.o' failed
make: *** [main.o] Error 1

scousepi, 2016-01-09:
Update on the compiling problem...I'm using the bcm2835 driver version 1.49 (rather than the 1.50 referred to in the documentation) and it now seems to be compiling and switching a test socket. Next step is to modify some code and see if it still works...

mark-c-pi, 2016-01-09:
Where do u get the bcm 1.5 drivers from?

scousepi, 2016-01-10:
There's a link provided in the Energenie documentation to instructions to download the 1.5 driver here https://gist.github.com/annem/3183536 I have since downloaded the v1.49 drivers which are allowing me to compile the example code with no problems. I've now crudely modified the example code and can successfully switch multiple sockets on and off. The only problem I've had is that I don't have the programming skills to programmatically create different sets of address bits and record them so I've hard coded them into different versions of the main.c source, with each different address operating 4 different MIHO002 SOCKETS.

LibanO, 2016-01-11:
Hi Dave,

I’ve also figured out why you we’re able to compile earlier. You’ll need to installed the older BCM library (1.37) attached.

Let me know how you get on.

Regards,

Liban Olaad
Development Engineer  MEng Hons MIET
[Energenie]<http://www.energenie4u.co.uk>     [cid:image0f393a.GIF@8e732942.458e7e61] <https://mihome4u.co.uk/>
tel: +44 (0) 1279 422022
Energenie, Unit 5, Harold Close, The Pinnacles, Harlow, Essex, CM19 5TH
energenie4u.co.uk
<http://www.energenie4u.co.uk>[Facebook]<https://www.facebook.com/energenie/>[Twitter]<https://twitter.com/#!/energenie4u>[LinkedIn]<https://www.linkedin.com/company/energenie>
Energenie is a trading name of Sandal Plc. Registered 03206855 in England. Claremont House, Deans Court, Bicester, Oxon, OX26 6BW

This communication is confidential and may also be legally privileged. Please notify us immediately if you are not the intended recipient. You should not copy it, forward it or use it for any purpose or disclose the contents to any person unless you are the intended recipient or are authorised to disclose. Internet communications are not secure and therefore Sandal Plc do not accept legal responsibility for the contents of this message. Although Sandal Plc operate anti-virus programmes, we do not accept responsibility for any damage whatsoever that is caused by viruses being passed. Any views or opinions presented are solely those of the author and do not necessarily represent those of Sandal Plc. Replies to this email may be monitored by Sandal Plc for operational or business reasons. Please ensure any anti-SPAM system you have in place is configured to allow emails from us. This may require you to contact your ISP.

Registered Office: Claremont House, Deans Court, Bicester, Oxon, OX26 6BW

Company Number: 03206855 (Registered in England and Wales)

scousepi, 2016-01-12:
Hi Liban,

Thanks for the information. I've now got everything working (with the MIHO002 switches anyway) using bcm library version 1.49 as 1.37 isn't supported on the Pi2. It compiles the TxDemo program and I've hard coded sets of different address bits. I'm currently switching 10 different devices mixing MIHO002 and ENER002 sockets. Apart from some transmission range problems it seems to be working. I've now modified the code so it will support up to 40 sockets. It's not an elegant solution but it's working for me at the moment. 

Next job, when I've got some more cash it to get hold of a MIHO005 and try working with that and the two-way communication.

Do you know if it Is possible to increase the transmitter output power? I'm currently assuming it's not.

I really appreciate the help - after 3 weeks of trying, with your advice I had it all working in a couple of days!

LibanO, 2016-01-12:
Hi Dave,

The hardware is set to 13dB transmission output power by default. Not changeable I'm afraid

Regards,
Liban

loriscaren, 2016-01-12:
I too hit compile issues with bcm2835 v1.5. DONT USE IT. As the name suggests, it's much older than 1.37 which itself is outdated (and wont in my experience operate the SPI on a Pi2). I compiled against 1.49 (the latest at present) and it defines RPI_V2_GPIO_P1_13 and the other compile issues - and best of all allows the 314RT to work on a Pi2.

whaleygeek, 2016-01-12:
The only thing that it is missing is that it doesn't consult the device tree for the base address of the SPI peripheral. That was the change they made in the hardware when moving from BCM2835 to BCM2836, the peripherals moved around. So rather than code reading from an absolute base address which was unique to the older chip, newer code now reads from a device tree file, which is populated by the firmware to have the correct base address of the peripheral.

All of this is a bit academic anyway (well, for this particular repo), because the purpose of this repo is to provide a zero-configure zero-install python library, so that people don't have to muck about configuring and building code - which in my opinion is a waste of your valuable time if all you want to do is turn a switch on and off!

I've been talking to @LibanO about planning adding the OOK support to the python code (which is not too much work as it's mainly a bit of radio configuration and a bit of message formatting). I will do this soon I think.

whaleygeek, 2016-03-01:
This issue seems resolved now, closing.

closed 2016-03-01 by whaleygeek