whaleygeek/pyenergenie #21: Using the MiH0013 eTRV

loriscaren, 2016-01-12:
Using monitor.py I get decoded packets with nice temperature readings from my eTRV, but has anybody got the valve to open/close - and if so, how?  Is there a tech data-sheet available for it?

ninjawil, 2016-01-12:
Not sure about eTRV but should be the same. The code below is from monitor.py toggles the output every 60seconds.

```
if sendSwitchTimer.check() and decoded != None:
            request = OpenHEMS.alterMessage(SWITCH_MESSAGE,
                header_sensorid=decoded["header"]["sensorid"],
                recs_0_value=switch_state)
            p = OpenHEMS.encode(request)
            radio.transmitter()
            radio.transmit(p)
            radio.receiver()
            switch_state = (switch_state+1) % 2 # toggle
```

I have forked this code and started to make it object orientated for a project of mine. It's basically monitor.py but rearranged so it can be used in other scripts. I removed the while loop and added the send data function. You can also just run the code and it will access the device, grab data, write it to the CSV file and send instruction to the device. I have tested it with the MIH0005 and ener314rt.

Please feel free to have a look and try it out. See:

https://github.com/ninjawil/pyenergenie

The code below will set up a device, grab the data from it, write it to the CSV

```
import pyenergenie.src.ener314rt as ener314rt

plug = MiPlug()

try:
    data = plug.get_data()
    plug.updateCSV(data)
    plug.send_data(True)

finally:
    plug.close()
```

whaleygeek, 2016-01-12:
I'm just confirming against the eTRV design spec with @LibanO, but if you look at the OpenThings specification, temperatures are parameter type 't'. A report has bit 7 clear (0x74) and a command has bit 7 set (0xF4) as per the OpenThings protocol.

I'm also confirming with @LibanO what data type encoding that command parameter will need to be recognised and processed as a temperature setpoint by the eTRV device,. But I'd hazard a guess that it's the same data type as returned by the 't' (0x74) report. 

Devices in OpenThings can support a whole range of data encodings, but usually an embedded device will only decode one particular format, and sending it the wrong format will either be rejected, or just interpreted incorrectly.

I know Liban has access to this information, but if you are getting impatient, you could reverse engineering it by 'just trying the above and see if I am correct'.

loriscaren, 2016-01-13:
Temperature packets are automatically emitted from (my) eTRV every 5mins and Liban tells me that it then goes to sleep (ignoring radio traffic). So, what I really want to do is provoke some other reply or (better) valve activity. So far I've not managed even by changing the tx in monitor.py from a JOIN_ACK to an ETRV_SEND_BATTERY_LEVEL. I'm guessing that other mains powered devices can afford to run their radio for rxing all the time?

whaleygeek, 2016-01-13:
Ok yes, good point. I'll wait to hear back from my request to Liban to see what the design spec for the device says. There will be a way to do it, but I'd rather understand how it was designed to work, than trying to reverse engineer it - reverse engineering always leads to a brittle solution based on false assumptions, in my opinion.

loriscaren, 2016-01-15:
Have managed to provoke valve movement and get a battery voltage returned.  The trick is to wait for an rx to arrive and then quickly do a tx before the TRV slumbers again. Liban sent me several somewhat confusing documents concerning the eTRV but I enclose a list of the commands I have managed to get to work.
[messages.txt](https://github.com/whaleygeek/pyenergenie/files/92094/messages.txt)

closed 2016-01-15 by loriscaren