whaleygeek/pyenergenie #22: Can't decode payload:'bad CRC'

LibanO, 2016-01-27:
receiving payload
readfifo:['0xd', '0x16', '0x8e', '0x77', '0x2f', '0x64', '0x26', '0x22', '0x6a', '0xc8', '0x8c', '0x6e', '0x7e', '0xb3']
0xd 0x16 0x8e 0x77 0x2f 0x3a 0xc6 0x6 0xb2 0x2f 0xda 0xd1 0xd0 0x39
Can't decode payload:'bad CRC'
receiving payload
readfifo:['0x3d', '0x29', '0x77', '0x85', '0x1e', '0x36', '0xcf', '0xab', '0x93', '0x5', '0xbc', '0x33', '0x65', '0xf8', '0x1d', '0x33', '0x98', '0x3', '0xa0', '0x1d', '0xbf', '0xf9', '0x34', '0xc6', '0xa2', '0xf1', '0xb6', '0x5a', '0x8', '0x45', '0x7f', '0x61', '0x5d', '0x95', '0x55', '0xf4', '0xaf', '0x8d', '0xd5', '0xbd', '0xa1', '0x22', '0x1c', '0x98', '0x2f', '0xbf', '0x6f', '0x3c', '0x2a', '0x2f', '0x68', '0xa1', '0x5f', '0x4a', '0xc8', '0x38', '0xb1', '0x89', '0xe0', '0xf2', '0x72', '0x76']
0x3d 0x29 0x77 0x85 0x1e 0x65 0x63 0xd8 0x4e 0xc9 0xdc 0x53 0xe5 0x77 0xbf 0x6c 0x29 0x41 0x60 0x50 0x32 0x42 0x8 0x2c 0x60 0xc5 0x43 0x10 0x57 0x84 0xeb 0xb1 0x2d 0x50 0x53 0x54 0x61 0x7c 0xe5 0xba 0xa8 0x67 0x36 0x6c 0x8c 0x69 0x4c 0x36 0x97 0xd0 0x84 0x10 0xb5 0x98 0x31 0x77 0x25 0x41 0x72 0xd9 0xe7 0x47
Can't decode payload:'bad CRC'
 
Any idea what could be causing this?
Regards, 

Liban

whaleygeek, 2016-01-27:
A couple of us have seen this from time to time, we think it is because another (non Energenie) device is transmitting on the same frequency. Because it does not have the Energenie encryption seed in the message, when the message is decoded the CRC is incorrect.

The way to verify this is to turn off all your energenie sending devices, and see if you still get the message. If you do, then don't worry about it. You can reconfigure the code to prevent it displaying these bad CRC messages by silently dropping them, if this is the case.

whaleygeek, 2016-01-27:
To disable this message, comment out this line in monitor.py

https://github.com/whaleygeek/pyenergenie/blob/master/src/monitor.py#L164

whaleygeek, 2016-01-27:
To turn off the trace messages in the radio module, replace this line with a 'pass' statement

https://github.com/whaleygeek/pyenergenie/blob/master/src/energenie/radio.py#L17

whaleygeek, 2016-01-27:
To stop the dumping of OpenHEMS messages, comment out this line:

https://github.com/whaleygeek/pyenergenie/blob/master/src/energenie/OpenHEMS.py#L161

whaleygeek, 2016-01-27:
Also some chatter in #19 that is related to this issue about bad CRC

whaleygeek, 2016-03-01:
This issue is *either* due to a non Energenie device in the near vicinity, or a device that is out of range (and therefore causing a corrupted packet). Closing as #19 deals with this.

closed 2016-03-01 by whaleygeek