whaleygeek/pyenergenie #23: Simple On/Off commands?

PRich13, 2016-02-02:
I realise this may be a sideline to this project but I've exhausted all other means.

I have the ENER314-RT and two simple on/off wall sockets, the ENER002.
(https://energenie4u.co.uk/catalogue/product/ENER002-2PI-RT)

I cannot for the life of me get these to work outside of their example C script.

All I want to do is be able to turn them off and on with a few simple commands which I can then start to build into larger projects (web server etc). I'm just starting my coding adventure and this has really started to sap my enthusiasm.

Any pointers / nudges in the right direction would be awesome. I'd be more than happy to feed back any progress to the community.

Like I said, sorry if this isn't the right place. I'm new to all this and this seems the most comprehensive body of work on the ENER314-RT as the other gits seem to fork from it.

PatchworkBoy, 2016-02-02:
I've got some c scripts for s1on/s1off & s2on/s2off I can fire over...

PRich13, 2016-02-02:
that would be an great starting point. 

See this is why I love the internet, peoples are just awesome.

PatchworkBoy, 2016-02-02:
Just edited my post with the link - if they don’t work I’ll be back online tomorrow night...

You need to have followed the ReadMe.txt in ENER002/HopeRF_TRX(ENER002) first to install the bcm drivers etc (from the software download on Energenie site)

PRich13, 2016-02-02:
Thanks man,

I'll try things out now and see what happens. I have all the drivers etc set up and I was trawling through the various files in the ENER002 folder and trying to modify things to work but didn't have much success.

I'll report back with my findings

PatchworkBoy, 2016-02-02:
They're not the most user friendly start to coding, I have to admit!

PRich13, 2016-02-02:
tried running with:

sudo ./s1on 

but getting command not found? Obviously I've opened the terminal from the folder location. Even tried copying them into the ENER002 folder and I can run the hoperf_trx but not yours.

I'll have a look at the app_main.c and see if I can start getting my head around that

PatchworkBoy, 2016-02-02:
Copy the HopeRF_TRX(ENER002) folder 4 times... once for each command you need to make. Then edit each app_main.c accordingly, edit the Makefile with your new binary name, and compile. 

Will post a sample around this time tomorrow, when I'm not in bed on my phone!

PRich13, 2016-02-02:
Thanks, I've always depended on the kindness of strangers.

whaleygeek, 2016-02-02:
The purpose of this work was/is to develop a (almost) pure python building block that people could use without installing anything and without having to build code. For the devices I have implemented to date, it works out of the box.

The ENER002-2PI-RT devices use a slightly different radio configuration (OOK modulation rather than FSK modulation, and some different messaging structures). They are not, as yet, supported by this python library. So, for the short term at least, you've got to use the C code (which is a real pain to use).

However, there's not much work required to get the OOK devices working in this same library: I'm in talks with Energenie at the moment about extending this python library to support the older green-button OOK devices as well. 

In the meantime, persevere with the code that @PatchworkBoy has sent you as it might get you started.

You could also take a look at this package which was developed by Amy and Ben and Gordon specifically to work with the older devices with the green button on them, as it might be of use too: 

https://pypi.python.org/pypi/energenie

PatchworkBoy, 2016-02-02:
If memory serves pypi supports the ener314 (GPIO based) not the 314RT (SPI based)..? can't recall offhand so I could be wrong. Either way, worth a try!! 

Problem at the moment is the 314RT is currently being sold bundled by energenie with the older green button ENER002 sockets, whilst the provided c software being FSK focussed & lack of documentation means no-one gets very far fast.

PRich13, 2016-02-02:
Hi, Thanks!

I've already tried using that package but it seems to be tailored to the older ENER314 (no RT) and certainly doesn't produce any results that I have experienced. 

Thanks to @PatchworkBoy I now have two scripts in C which turn the sockets off but as of yet I haven't figured out how turn them on. I'm pleased enough so far considering I don't know a letter of C (or any language for that matter).

I look forward to the possible implementation of this functionality within this project and will gladly test anything.

PatchworkBoy, 2016-02-02:
(@Liban should really get together with @gizmocuz and sort Domoticz support - that would immediately position the 314RT as a competitor to the RFXCom RFTRX433e for a quarter of the price... but I'm digressing offtopic - sorry!)

PatchworkBoy, 2016-02-03:
Righty @PRich13 - see app_main.c, dev_HRF.c and Makefile in the HopeRF_TRX(ENER002) folder. 

Note relayState definition in app_main.c & dev_HRF.c  - this is what determines the command to send (dev_HRF.c contains the actual bits, app_main.c just references)

https://github.com/PatchworkBoy/basics-ener314rt/

PatchworkBoy, 2016-02-03:
NB: You need to make sure the sockets are paired...

Stick one in learning mode and send s1on to get it to become aware that it’s socket1. Repeat for the 2nd, but sending s2on to get it to learn that it’s socket2.

PRich13, 2016-02-03:
Thanks @PatchworkBoy I'll start experimenting with this now. 

will the sockets need pairing after each reboot?

PatchworkBoy, 2016-02-03:
Nope. Sockets should remember pairing indefinitely... unless there’s a power cut... then they may forget (not had one yet to find out!)

PRich13, 2016-02-03:
currently experimenting with trying to generate the codes for sockets 3 and 4 as well as all on and all off. I have what I think is the right code. All on and off works but when trying to teach my current socket 2 as 3 the flashing light stops when the script is run but it doesn't seem to learn it.

I can upload the code if curiosity takes you

PatchworkBoy, 2016-02-03:
Aye post it up... I can add it into the repo (or you can fork the repo for yourself and add it into that)

PRich13, 2016-02-03:
Still figuring out how to post to git so for now:

// THE MAGIC LINES
	if (relayState == 0)
	{
		printf("Socket 1 ON\n\n");
		buf[15] = 0xEE;		// D0-high, D1-h		// S1 on
		buf[16] = 0xEE;		// D2-h, D3-h
	}
    if( relayState == 1)
	{
		printf("Socket 1 OFF\n\n");
		buf[15] = 0xEE;		// D0-high, D1-h		// S1 off
		buf[16] = 0xE8;		// D2-h, D3-l
	}
	if (relayState == 2)
	{
		printf("Socket 2 ON\n\n");
		buf[15] = 0x8E;		// D0-l, D1-h		// S2 on
		buf[16] = 0xEE;		// D2-h, D3-h
	}
	if (relayState == 3)
	{
		printf("Socket 2 OFF\n\n");
		buf[15] = 0x8E;		// D0-l, D1-h		// S2 off
		buf[16] = 0xE8;		// D2-h, D3-l
	}
	if (relayState == 4)
	{
		printf("Socket 3 ON\n\n");
		buf[15] = 0xE8;		// D0-h, D1-l		// S3 on
		buf[16] = 0xEE;		// D2-h, D3-h
	}
	if (relayState == 5)
	{
		printf("Socket 3 OFF\n\n");
		buf[15] = 0xE8;		// D0-h, D1-l		// S3 off
		buf[16] = 0xE8;		// D2-h, D3-l
	}
	if (relayState == 6)
	{
		printf("Socket 4 ON\n\n");
		buf[15] = 0x88;		// D0-l, D1-l		// S4 on
		buf[16] = 0xEE;		// D2-h, D3-h
	}
	if (relayState == 7)
	{
		printf("Socket 4 OFF\n\n");
		buf[15] = 0x88;		// D0-l, D1-l		// S4 off
		buf[16] = 0xE8;		// D2-h, D3-l
	}
	if (relayState == 8)
	{
		printf("All Sockets ON\n\n");
		buf[15] = 0xEE;		// D0-h, D1-h		// All on
		buf[16] = 0x8E;		// D2-l, D3-h
	}
	if (relayState == 9)
	{
		printf("All Sockets OFF\n\n");
		buf[15] = 0xEE;		// D0-h, D1-h		// All on
		buf[16] = 0x88;		// D2-l, D3-l
	}
	// END MAGIC LINES

PatchworkBoy, 2016-02-03:
Cool... Ta! Will get those added in to dev_HRF.c

PRich13, 2016-02-03:
Like I said, not having much success pairing for 3 and 4 yet

hotear, 2016-02-29:
Hey there whaleygeek, I would like to request this functionality (you mention above) be added to the python code.  I have a couple of spare green style plugs that I am no longer able to use since swapping to the new transceiver board and this (great) python only code for reading the purple style plugs.  It would be good to make use of them again as they are much cheaper than the purple ones and I don't need to know the power draw from all devices, just to be able to turn them on and off.

whaleygeek, 2016-02-29:
I'm designing the OOK support at the moment, which will then support the legacy devices. When this is done and tested, I'll release it here and you can clone it and try it for me if you like :-) Thanks!

PRich13, 2016-02-29:
Second Tester ready and waiting! I'm having a few stability issues with the customised C scripts.

whaleygeek, 2016-03-17:
I've just added OOK support, and have a very simple test script working. I am closing this issue as it is now tracked in #32 please see the instructions in that issue on how to try out my test code, and read carefully the warnings about it being 'first working code' and 'lots more to do to it'. As they say, release early, release often :-)

closed 2016-03-17 by whaleygeek