whaleygeek/pyenergenie #31: New feature requests

whaleygeek, 2016-02-29:
To all the fork's, stargazers and watchers of this repo.

I am reviewing the status of this code and a number of outstanding feature requests, and I just wanted to put out a call for anything else that people would like to see in this code.

I am aware that there are some pull requests already, and that some of the forks have started their own innovative paths - but I just wanted to ask anyone that is interested in this code moving forward, what things they would like to be able to do with it in the future.

Please comment against this issue, and I will collate feedback. I can't promise to implement everything that is requested, but anything that you feed back via this mechanism will be considered in the wider scheme of things.

David
@whaleygeek

OpinionatedGeek, 2016-03-06:
Hi there,

First of all, thanks for the code! I couldn't get anything really working on my Pi2 and MIHO005 apart from your software.

Things I'd like to see:
* A simple command to turn a device on or off (I have a pull request that does this but it's more a copy&paste than a good example of code re-use. I won't be offended if  you don't use it.)
* A simple command to tell if a device is on or off. (As long as the device supports this, of course.)

I like the idea of these simple commands because they can then be composed into bigger scripts. For instance, I like the idea of giving friendly names to devices, so I could call:

  turn_on downstairslight

Something would then map the 'downstairslight' to device ID 0x123 and call the appropriate code to turn on device ID 0x123.

This mapping of names to device IDs needn't be part of the pynergenie code - it could be separate as long as there were simple commands for it to call to do the actual device switching.

Another thing I'm a little interested in is monitoring the power used by a MIHO005. I don't have any plans around this, but I can see how graphing power usage, or calculating the cost (given a per kwh figure as input?) might be useful. The monitor.py already does some of this, but it seemed a more general monitoring program than this specific use case.

I hope this perspective is useful - and of course feel free to ignore it all!

whaleygeek, 2016-03-06:
Sounds great, thanks! Yes, what you are talking about there in your first point is a bit like DNS (domain name system) but for Energenies - you want to refer to them by a friendly name rather than as their device address. I've been thinking through some ideas for a device registry, so I'll follow up your ideas in that one.

The simpler turn on/off - yes, on the list, definitely. The code I wrote was originally just to get the thing working, so it focuses on the messaging and the radio interface. I want to add support next for the older legacy green button devices, then think a bit higher up the stack and provide more useful services that others can build applications from.

Also, I'm hoping to build some switching/monitoring example programs, once the higher level simpler application interfaces are done. I didn't want to waste time building examples at this stage, when I knew I wanted to build a higher more abstracted interface to the devices first.

Thanks for your input, and please do be a beta tester for me as new stuff becomes available!

The pull requests are useful thanks, as are all the forks. I do look at them all and everything that everyone else has done with this code has influenced my thinking on the way forward.

whaleygeek, 2016-05-13:
Split out new feature requests into #53 and #54. Closing.

closed 2016-05-13 by whaleygeek