whaleygeek/pyenergenie #38: NodeRed node development

whaleygeek, 2016-04-16:
I would like to develop some NodeRed nodes for each of the different Energenie products. 

This will make it possible to wire up NodeRed flows using any Energenie device.

I plan to use the same integration scheme that RPi.GPIO nodes use inside NodeRed on the Raspberry Pi - i.e. NodeRed Node calls into the python/C code.

cutec-chris, 2016-10-30:
Any news on this ?

whaleygeek, 2016-10-30:
No specific progress on this, due to being busy with other projects. It is still on the radar, but no timescale defined, sorry.

cutec-chris, 2016-10-30:
havaned tried it now but...
https://github.com/DaveThw/rpi-energenie

whaleygeek, 2016-10-30:
Thanks @cutec-chris - that uses the (simpler) GPIO driven radio (one way radio). The code in this repository is more focused on the more complex two-way radio, which works with every Energenie device ever manufactured, including the whole MiHome range (for energy monitoring).

But the link will be useful I think, thanks!

Achronite, 2017-11-17:
Any news on this?  I have the ENER314 working with GPIO and Node-red, but I've just bought a MiHome and ENER314-RT and I need to upgrade my node-red flows.

whaleygeek, 2017-11-17:
Sorry, not at the moment, got a bit busy with other projects. Although you should check in with AndySC on twitter, he does lots of NodeRed stuff and often builds on top of some of my project ideas, so he might have done something with it already.

gpbenton, 2017-11-20:
You could use my [ENER314-RT - MQTT ](https://github.com/gpbenton/engMQTTClient)project and use the Node-Red MQTT node.

Achronite, 2019-01-12:
I'm currently working on this.  I have a new node red-node node working that controls all of the OOK based devices, it should be released soon.

I've no plans to do the monitor device though, as I don't own any of them.

Achronite, 2019-04-28:
My node-red nodes are now released into beta. The nodes caters for OOK and FSK devices, with switching & monitoring, and use @whaleygeek C code from this library.  I've rewritten all of David's python routines into javascript or C so that they integrate better with node-red.

You can find the nodes here:
[npm](https://www.npmjs.com/package/node-red-contrib-energenie-ener314rt)
[github](https://github.com/Achronite/node-red-contrib-energenie-ener314rt)

All testing and comments welcome!

whaleygeek, 2019-04-28:
Looks good. Thanks for correctly attributing the copyrights to my base work (you'd be surprised how often people don't!!)

whaleygeek, 2019-04-28:
ps. The C code is mine too. I used the original C from Energenie as a benchmark, but rewrote it all from first principles. So I don't think anything in my repo came from Energenie at all. next time you update perhaps you could tweak the authors lists as appropriate. I put it in a folder called energenie because it supports their devices.

This is cracking work, well done.

Achronite, 2019-04-28:
Will do. I'll stage it now in my develop branch :-)

whaleygeek, 2019-05-05:
This work is now published, and written by a member of the community - nothing for me to do here :-) Thanks @Achronite for the excellent work!

closed 2019-05-05 by whaleygeek