whaleygeek/pyenergenie #44: Python 3 support

whaleygeek, 2016-04-20:
I haven't tested any of this with Python3 yet. Related to discussions recently with Nicholas Tollervy from PSF, bringing this up to date to support Python3 would be a good thing. I'd like to still support some Python 2 versions though, for those who have applications written in 2 that they are not able to easily port to 3 for other reasons.

It's possible it won't work in Python 3 due to package import rules changing, and use of raw_input, at least.

AndrewNRoss, 2016-04-20:
I've tried the last stable version with my green button plugs and python 3. It works, but it needs some minor changes to the import rules and the use of raw_input, but nothing else. 

I think it should be possible to support both with only relatively minor changes. If you are happy with supporting 2.5+ only, then using the __future__ import will make python 2 behave like the python 3 with regards to relative imports within a package.

Python 3 renamed raw_input to the more sensible input. Google shows a number of kludges to get code that works with python 2 and 3. This is only really in the test code anyway rather than the core library, so it might matter less.

whaleygeek, 2016-05-16:
I supported Python 3 with a few simple changes:

rawin() is defined as raw_input or input appropriately in legacy.py which is the only module that uses it.

all dict.has_key() references changed to key in dict.

'except ImportError' put around the few relative imports that needed fixing.

range(int(number/2)) used in the for loop that failed because it thought it was a float.

Tested with both python 2.7.9 and python 3.4.2 on Raspberry Pi Raspbian Jessie on a Pi B+

Closing.

closed 2016-05-16 by whaleygeek