whaleygeek/pyenergenie #59: Home Brewing application

Jeremy13ullock, 2016-05-26:
Hi David, 

I'm building an application to monitor the fermentation progress of my home brewed beers in two fermentation chambers.

I'm using temperature sensors to monitor beer temperature and the Energenie switch to control the heating/cooling of those chambers.

I'm now at the point where I'd like to be able to read back the status of the Energenie switches. Do you have any pointers as to how i can do this.

Regards

Jeremy

whaleygeek, 2016-05-26:
Sure, could you let me know the model numbers of the sockets you wish to monitor, as I need to confirm what type of devices you are using before I can give any specific advice that might help. Thanks!

whaleygeek, 2016-05-30:
@Jeremy13ullock Note that the device_classes sprint has completed and the code is now merged to master - you might find some of the new demo programs a good starting point for your brewing application, not least because there is a mihome_energy_monitor.py example that is close to what you want on the receive side, and all devices are now easy to control by putting them into a registry file (using setup_tool.py or the join discovery behaviours as per discover_mihome.py) and then controllable with name.turn_on() and name.turn_off().

Let me know how you get on with this.

I'm closing this issue because I think the latest code should give you everything you need to get going, but do open it again if you have any outstanding issues not resolved by this new release of code.

closed 2016-05-30 by whaleygeek