whaleygeek/pyenergenie #72: handling of bad payloads for FSK receive

whaleygeek, 2016-06-03:
There is some evidence that when an FSK payload gets corrupted (especially from a device that transmits multiple message repeats), that the receive FIFO fills up with partial packets, and this prevents some future decodes.

There might be some 'hardening' of the receive processing logic possible here, to clear out the FIFO after a time limit if a partial packet seems to have been received. Will have to think about where in the receive pipeline this should be done for best effect, because the contents of the payload are not understandable until they have been decrypted and CRC checked.

jmigreen, 2019-04-28:
I find this occurs more often when the MiHome switch is off than when it is on. Anything about the hardware that you're aware of that explains this behaviour?

whaleygeek, 2019-04-28:
Hmm. Nothing specific that I can think of that would change the reliability when in OFF mode. I think the payloads are the same length too, so it's unlikely that we're hitting some strange modulus effect of the FIFO size.