whaleygeek/pyenergenie #75: mihome_energy_monitor power readings not working

whaleygeek, 2016-06-08:
Mark reported that the mihome_energy_monitor.py example reports:

Incoming from (4, 2, 911)
No route to an object, for device:(4, 2, 911)
message from unknown device:(4, 2, 911)

whaleygeek, 2016-06-08:
The issue here is that the example code iterates through the registry, but the registry is a list of devices that you own, not necessarily a list of devices that are connected up to your code.

To create an object that can store values, you have to call registry.get() which creates an instance of that object, and also knits up a receive route. Under the bonnet, the registry might already have a cached object for convenience, but it is not knitted up to the receive router (fsk_router), so when a message comes in from it, it will be logged as an unknown device.

The workaround at the moment is to make sure you 'get' it from the registry, which makes the object active. The examples control_any_auto, control_any_reg and control_any_noreg show various ways to do this.

I will look back at the energy monitor example and improve and correct it so that it causes the receive routes to be knitted to real objects.

whaleygeek, 2016-06-08:
e.g. if your device is in the registry named as 'tv', try something like this:

import energenie
import time

tv = energenie.registry.get('tv')

energenie.init()
while True:
  energenie.loop()
  p = tv.get_power()
  print(p)
  time.sleep(1)

hudsonkk, 2017-01-04:
Hi whaleygeek

I've just started learning python as part of my project to use energenie products around my home. I'm trying to generate a script that will just turn off the MIHO005 via the ENER34RT on my Pi.

I'm having difficulty in translating the code here into code that delivers what I need, specifically I'm receiving a keyError as shown below (I've tried a range of device and name combinations and looked into the kvs.py, devices.py and registry.py files to understand how the key is generated).

I hope it's ok to ask for guidance here, if so could you point me in the right direction please?

Code:

def off():

    key = "Monitor1 MIHO005(0xffff7a)"
    device = energenie.registry.get(key)
#    device ="Monitor1 MIHO005(0xffff7a)"    
#    device = auto_0x2_0xffff7a
#    print"Turning off %s" % device
    device.turn_off()

if __name__ == "__main__":

    energenie.init()
    try:
        off()
    finally:
        energenie.finished()

Error:
  File "/home/pi/pyenergenie-master/src/energenie/Registry.py", line 63, in get
    c = self.store[name]
  File "/home/pi/pyenergenie-master/src/energenie/KVS.py", line 103, in __getitem__
    return self.store[key]
KeyError: 'Monitor1 MIHO005(0xffff7a)'

whaleygeek, 2017-01-04:
If you already know the deviceid, add 3 lines to registry.kvs like this...

ADD tv
type=MIHO005
device_id=1675

Replace the 1675 with the code you got back from auto discovery from
running the setup.py and holding the button on the front for 3seconds. It
can be a decimal number like 1675 or a hex number eg 0x123456 but it must
be the number that was discovered by running setup.py discovery option and
pressing white button on front for 3 seconds.

Then in your code just say tv.on() and tv.off() and it should work. The
energenie.init creates named variables for any ADD that is in registry.kvs
note no spaces or special symbols allowed in the name.

hudsonkk, 2017-01-04:
Thanks for the guidance. I have this in the registry.kvs

```
ADD Mon1
type=MIHO005
device_id=16777082
```
I'm now getting the error NameError: global name 'Mon1' is not defined

The code is
```

import time
import energenie
##from energenie.lifecycle import *

#===== GLOBALS =====

quit = False

#===== INPUT METHODS ==========================================================

try:
    readin = raw_input # Python 2
except NameError:
    readin = input # Python 3

def off():
        Mon1.off()

if __name__ == "__main__":

    energenie.init()
    try:
        off()
    finally:
        energenie.finished()
```

I'm stumped.

whaleygeek, 2017-01-04:
I'll check tomorrow, but I think you have to call a registry function for
it to auto create the user variables. Will look in morning for you.

whaleygeek, 2017-01-05:
Ok, so the best thing to do is base it on one of the following examples,
depending how you want to work.

If you want a variable called 'tv', try control_any_auto.py

If you want to interact with the registry via a key, try control_any_reg.py

If all you have is a single socket and you want a really light and simple
direct solution, bypass the registry totally, and base your code on
control_any_noreg.py - looking at your example code you sent me, I suggest
you start with this. If you already know your device_id from the discovery
process in setup.py, then you can hand-code a single
energenie.devices.MIHO005() object with the known device_id in the
brackets, and interact with that directly.

import energenie
tv = energenie.devices.MIHO005(16777082)
energenie.init()
tv.off()
energine.finished()

The golden rule would be to choose the code starting point that (a) you
most understand and (b) fits more closely to the size and scale of your
application. The registry is an optional feature that you don't have to use
if you don't want to. It's more useful where you have a large number of
devices and want to loop through them all automatically reading data from
them etc etc.

If you want to receive data from devices as well, you'll need a loop, so
some of the example .py files in the top level directory that I refer to
will give you some design patterns to get started with.

Hope this helps, and let us know how you get on!

hudsonkk, 2017-01-05:
That's great advice - I just had to amend off to turn_off and capitalise Devices and it works!

```
import energenie

tv = energenie.Devices.MIHO005(16777082)

energenie.init()

tv.turn_off()

energenie.finished()
```
`
This is the right level for me and something that I can build on.

Thanks for your help with this.

whaleygeek, 2017-01-05:
Hi,

Great, glad you got it working! Thanks so much for reporting back.  Have
fun turning your TV on and off!!

Also, when your project is finished, please do blog about it or report back
and let us know what you built, it's always great to hear from people that
build real things based on this code.

David

whaleygeek, 2019-05-05:
Looks like this issue is now fixed so I will close, but please re-open if I missed anything @hudsonkk

closed 2019-05-05 by whaleygeek