whaleygeek/pyenergenie #80: Add the ability to record a house code from a RF hand remote

jiteshvassa, 2016-12-10:
Your readme doc alludes to this, just wondered how far along you were?

whaleygeek, 2016-12-10:
No progress, sorry, been too busy with other stuff. It needs a whole chunk of code added into radio.c to configure the RFM69 module as an OOK receiver, detect the start of payload, decode the payload by removing the physical layer encoding, and return a buffer with the house code in via the .py interface.

djthorpe, 2017-03-03:
Hi, I'd also be keen to do this (I'm writing my code in golang rather than python) but so far I've been unable to get the RFM69 to detect the start of the payload when in OOK mode. Here's the parameters I have been using, any hints as to the right parameters I should be using?

```
	case MI_MODE_OOK:
		if err := this.radio.SetModulation(hoperf.RFM_MODULATION_OOK); err != nil {
			return err
		}
		if err := this.radio.SetDataMode(hoperf.RFM_DATAMODE_PACKET); err != nil {
			return err
		}
		if err := this.radio.SetSequencer(true); err != nil {
			return err
		}
		if err := this.radio.SetFreqDeviationUint16(0); err != nil {
			return err
		}
		if err := this.radio.SetFreqCarrierUint24(0x6C7AE1); err != nil {
			return err
		}
		if err := this.radio.SetAFCRoutine(hoperf.RFM_AFCROUTINE_STANDARD); err != nil {
			return err
		}
		if err := this.radio.SetRXBW(hoperf.RFM_RXBW_MANT_24,hoperf.RFM_RXBW_EXP_4,hoperf.RFM_RXBW_DCC_4); err != nil {
			return err
		}
		if err := this.radio.SetBitrateUint16(0x1A0B); err != nil {
			return err
		}
		if err := this.radio.SetPreambleSize(0); err != nil {
			return err
		}
		if err := this.radio.SetPayloadSize(2); err != nil {
			return err
		}
		if err := this.radio.SetSyncKey(nil); err != nil {
			return err
		}
		if err := this.radio.SetPacketCoding(hoperf.RFM_PACKET_CODING_MANCHESTER); err != nil {
			return err
		}
		if err := this.radio.SetPacketFilter(hoperf.RFM_PACKET_FILTER_NONE); err != nil {
			return err
		}
		if err := this.radio.SetPacketFormat(hoperf.RFM_PACKET_FORMAT_FIXED); err != nil {
			return err
		}
		if err := this.radio.SetCRC(false,false); err != nil {
			return err
		}
		if err := this.radio.SetFIFOThreshold(30); err != nil {
			return err
		}
		if err := this.radio.SetOOK(hoperf.RFM_OOK_THRESHOLDTYPE_FIXED,hoperf.RFM_OOK_THRESHOLDSTEP_0P5DB,hoperf.RFM_OOK_THRESHOLDDEC_0); err != nil {
			return err
		}

```

whaleygeek, 2019-05-05:
@djthorpe sorry I missed this one. The OOK payloads need a preamble at the start, otherwise they are not detected by the switches. Also I discovered a long time ago (and fixed) a problem where if the radio is in the wrong state at the start of a transmit, it inserts a spurious 1/0 transition as the PA of the radio powers up.

The latest code seems to work great, and the comments there may help. I removed preamble generation from the radio driver, and generate it in the payload, as that avoided the 1/0 PA power up transition in the end.

Code here, with quite a few comments explaining how it works.

https://github.com/whaleygeek/pyenergenie/blob/master/src/energenie/TwoBit.py#L9

Sorry for delay in replying, it might be too late for you now, but hope this helps.

Closing but please re-open if you still have an issue you would like help with.

closed 2019-05-05 by whaleygeek