whaleygeek/pyenergenie #83: Light switches and wall sockets

rotahn, 2017-01-25:
Hi there, 
Looking at buying the pi board but I'm wondering if this library works with the lights and sockets.

Looking at the devices.py I see that there are no values against these.

If it's doesn't. ...could you advise how I would go about getting them..I don't mind playing around as long as it doesn't cost me money lol.

Cheers

whaleygeek, 2017-01-25:
The light switch (i.e. the wireable light switch that you use to replace your mains switch on the wall) and the green-button devices are all OOK (legacy) devices. Yes, it works with those.

The reason those device constants are not enabled in devices.py is that they all respond to a 'learnt' code as they are receive-only devices, so there is no such concept as a device code for them. You use the legacy learn mode in setup_tool.py and the devices are entered into learn mode, and they 'learn' the code that the Pi sends to them.

rotahn, 2017-01-26:
Ahh yeah of course...Feel a little stupid now lol.

I notice that you are/were going to take a look at getting this working in Domoticz directly. Did you get anywhere with this.

Have you thought about adding support for lightwaverf? I notice on github someone has done a C version for the pi, it uses GPIO. They use the OOK protocol also.

I'm just hopping onto the home automation bandwagon  sorry for all the questions.

whaleygeek, 2019-05-05:
Just reviewing tickets, sorry to miss this one!

Domoticz direct support is not really on my radar. I'm hoping that others with experience in that area will fork and get it working really.

Regarding lightwaverf - unlikely, as I am keeping this library entirely focused on Energenie products.

closed 2019-05-05 by whaleygeek