whaleygeek/pyenergenie #84: use of sudo?

warrenpaul, 2017-01-30:
Hi there, I'm trying to "simply" turn on and off some legacy green button energenie switches. I'm using HomeAsistant, which is running as a system user service on the raspberry pi, which then tries to call an energenie python script.

However, the system user (homeassistant) has no sudo privilege, and hence the scripts fails. Is there any way of executing the ENER002 device on/off command, using a script, but not using sudo? I don't mind manually passing the house code and device index to bypass the need for the registry if that's the problem.

your advice is much appreciated.

whaleygeek, 2017-01-30:
Yes, upgrade your pi to the latest os, it no longer requires sudo to access
gpio. Alternatively, find out what config changes the Raspberry Pi
foundation did to effect that change, and retro fit it to your existing OS.
Someone on the pi forums should know how to do this retro fit.

Hope this helps

gpbenton, 2017-01-30:
I am using HomeAssistant with my [mqtt client](https://github.com/gpbenton/engMQTTClient), if you prefer to use that rather than upgrade.  There is an example configuration in the readme page.

BobSammers, 2017-02-15:
> Yes, upgrade your pi to the latest os, it no longer requires sudo to access gpio
Is this the full story? I've got a recent Raspian (downloaded Jan '17) and the current version of the pyenergenie code (from the Whaleygeek repo, not the Energenie one) and I still need to use `sudo`. I get an error message, "can't open /dev/mem," otherwise.

I think the changes to Raspian introduced a new device `/dev/gpiomem` and pyenergenie needs to access the gpio through this device (which the `pi` user has group permissions on by default) rather than `/dev/mem` in order for the new improved permissions to work.

AIUI, the new device is masked off so that rather than giving non-root access to the whole memory space (which I think is what `/dev/mem` represents), the new device allows only access to the portion of the memory space where the memory mapped gpio locations are, making it a lot safer to offer to non-root users.

warrenpaul, 2017-02-15:
Hi there. My bad, I should have updated.... Bob, I had the same issue (/dev/mem), and some futile googling led me to conclude that the security issue was bigger than me.
I solved this by creating a 777 folder which home assistant could write to, and used a command line switch and a touch command to write a file (switch_id.on / switch_id.off), and wrote a polling script to poll that directory and fire off the appropriate pyenergenie command and delete the file. I just added the polling script to the pi startup.
I'm very happy to share my code, but to set expectations, its some time since I wrote sh shell (I'm not there with python), and there is no error handling (one day I'll tidy it up)... all that said, it works like a dream! :o)

BobSammers, 2017-02-15:
Thanks for the offer Warren. I don't have this problem specifically (my problem, such that it is, is simply that my cron and node.js scripts are running as root, which I'd rather not be doing). Someone else may find it useful, of course, but I am hoping that Whaleygeek will be able to update without too much hassle.

These lines (gpio_rpi.c):
```C
   /* open /dev/mem */
   if ((mem_fd = open("/dev/mem", O_RDWR|O_SYNC) ) < 0) 
   {
      printf("can't open /dev/mem \n");
      exit(-1); //TODO return a result code
   }
```
are the only point where `/dev/mem` is mentioned, as far as I can tell, so hopefully (and by someone who understands the rest of the `if` statement!), it will be straightforward to check for the presence of `/dev/gpiomem` and open that instead, solving my problem and removing the need for your (and any other) workaround.

BobSammers, 2017-02-15:
In fact, I've just confirmed that the following works in my setup (replace the code above in gpio_rpi.c):
```C
   /* open /dev/gpiomem (preferably) or /dev/mem */
   if ((mem_fd = open("/dev/gpiomem", O_RDWR|O_SYNC) ) < 0)
   {
      if ((mem_fd = open("/dev/mem", O_RDWR|O_SYNC) ) < 0)
      {
         printf("can't open /dev/gpiomem or /dev/mem \n");
         exit(-1); //TODO return a result code
      }
   }
```
I've tested that it falls back to `/dev/mem` if `/dev/gpiomem` cannot be found (i.e. an older version of Raspian), which then needs `sudo`, of course.

To update manually, there is a `build_rpi` script in the C code folder that will rebuild the C library accessed by the Python scripts and then you should be able to run your scripts without `sudo`. Hopefully Whaleygeek can look over the code and update the repo if it does, in fact, do all and only what it's supposed to.

whaleygeek, 2017-02-15:
@BobSammers looks fine to me. Are you able to send a pull request against this repo, and I'll test and integrate from there, and I'll update the default pre-built .so file with it, thanks :)

benjimeistro, 2017-08-09:
I would also like to confirm that this works as expected for me, e.g without sudo/root once the changes have been added to the existing code. I simply changed the existing entry in `gpio_rpi.c` from `"/dev/mem"` to `"/dev/gpiomem"` and then rebuilt the C library following the tip from BobSammers above, using the `./build_rpi` script. and everything is working great. I can integrate it into openhab now,  thanks for your time and hard work! :)

whaleygeek, 2017-08-09:
Super, thanks @benjimeistro for reporting back. I haven't had the time yet to merge and test this myself, but glad to hear that it all works. I will get round to merging this in and rebuilding the pre-built library soon I hope, but until then, the referenced pull request is useful for others to make this fix themselves.

Glad it all works - I haven't come across openhab before, if you have any links to reference a community /project/repo etc please do send them through here, as it helps others to find related projects. I plan to do a bit of a mop-up pass soon and update the master README with links to other projects, so having links to these other things that are going on as a spin-off from here will be great!

whaleygeek, 2019-05-05:
Noted that this is fixed by https://github.com/whaleygeek/pyenergenie/commit/4bc3f7d529428ecc9daa92cd7d653ec1757a3537

closed 2019-05-05 by whaleygeek