whaleygeek/pyenergenie #85: Adding more than 4 OOK based units with an ENER314-RT

catret, 2017-01-31:
Hi,

like the title says really , is there a way to add more than 4 of the 'legacy' units (socket adapters, light switches etc) with the ENER314-RT board?

I can't believe Energenie expect us to have no more than 4 light switches in the house so feel i must be missing something obvious.

thanks

whaleygeek, 2017-01-31:
Yes, you can change the house code assigned in the Python, and 'learn'
different house codes to different legacy devices. Then just register those
new devices in your registry with a different house code, and refer to them
by their variable name in the python.

So, use setup_tool/legacy learn option and enter a different house code to
the default.
Put the switch into learn mode
trigger the transmission from the setup tool with 1 2 3 4 as the device
code for that house code.

The switch has now learnt a different house code. So with 20 bits of house
code and 2 bits of device code, 12 bits of address will give you 2^22 =
4,194,304 unique devices. That should be enough for any large house ;-)

You can then store the combination of the house code and device code in
your registry.kvs

ADD fan
type=ENER002
device_id=[0x6C6C6, 1]

0x6C6C6 is the default house code - change this to a different house code
to get more address space.
1 is the device index (0 for all in that house code, or 1 2 3 or 4 for a
specific device).

If you don't want to use the registry, use the file
https://github.com/whaleygeek/pyenergenie/blob/master/src/control_any_noreg.py
and put the house code and device code inside the device initialisation,
e.g.

fan = energenie.Devices.ENER002((0x1234, 1))

You can then use fan.turn_on() and fan.turn_off() in your code.

Don't forget to use energenie.init() at the start and energenie.finished()
before script exit.

House codes are 6 hex digits (4 bits per digit=20 bits)

catret, 2017-02-06:
That worked a treat. many thanks

closed 2017-02-06 by catret