whaleygeek/pyenergenie #87: switch.py keeps radio in use after it's finished.

lukew151, 2017-02-22:
I have both legacy green button switches & newer purple button miHome adapters.  There is a difference in behaviour between the two programs  legacy.py and switch.py which operate my two different types switches:
legacy.py “closes” the radio correctly when it exits.
switch.py doesn't “close” the radio.  It was obvious using it that the LEDs remain on, even when you exit the program loop.   I later discovered that should you exit the loop while the program is transmitting, it continues to transmit – jamming everything in my flat which uses 433MHz!

Solution (I'm no coder,  I just tried something and it fixed my issue, but I'm unsure as to possible side effects!):  I have added a line to /drv/radio.c at 522:
`radio_standby(); `
so it tucks into the function radio_finished(void).  This appears to take the radio out of transmit/receive mode gracefully.

whaleygeek, 2017-08-09:
Sorry for the delay in replying and thanks for the fix, that was an oversight on my part! I'll roll this into the next set of changes, so I'll leave this issue open until I get round to doing that, thanks!

whaleygeek, 2019-05-05:
Waiting for @Minimad-Diver to confirm if this works with the gpio_finished after the radio_standby, as that seems the more obvious way round to do it. Please send me a PR once you confirm that and I'll merge to master, thanks!

whaleygeek, 2021-04-14:
I have recently added energenie.cleanup() that rolls up the GPIO closures into a nice function.

However, this still needs to be retro-fitted to all demo scripts.