whaleygeek/pyenergenie #88: MIHO026 & ENER314-RT

Lorccan, 2017-04-23:
I am probably being a bit thick but I am stuck on what to do next:

I can successfully get setup_tool.py to connect to the MIHO026 and then to repeatedly switch it ON and OFF, but I don't see anything written to the registry.kvs file and I don't know where I get the data from the learning process to put in it.

Could you elaborate a bit please on how you go about adding a new device to the setup?

Lorccan, 2017-04-23:
Got it now, but it could be clearer!

whaleygeek, 2017-04-23:
Glad you fixed it!

If you let me know what the point of confusion was, and what you did in the end to fix it, I'll gladly review that part of the documentation based on your feedback.

Lorccan, 2017-04-24:
It was really a gap between getting the device to switch on and off and the fact that the default registry contains a reference to all current types of OOK device (and that its the switch that learns which transmitter is sending to it - rather than the transmitter than learns the device).  In my case, I only need to use this one type, so I edited the registry down to just that entry.

My next challenge is that I want to use this with Domoticz, but I am finding that a very cut down .py that works fine from the command line (I have separate ON and OFF scripts) does not work when called from either .sh or .lua scripts.  I suspect it's a dependency/scoping issue, but I haven't worked it out yet. Did you go any further with Domoticz integration?

Lorccan, 2017-04-25:
I'm reopening, because I am not sure you are alerted to responses when the issue is closed.

whaleygeek, 2017-04-26:
Thanks @Lorccan - spotted it now, thanks.

No further with Domoticz, I didn't really get any responses to my posts on the forum (or if I did, perhaps it didn't send me notifications?)

The main thing will be that it accesses /dev/mem, so unless you do the hack described in some other ticket here for the gpiomem sub range of protected memory, it will need to run as root.

If it's a permissions thing, you can use the setuid trick to orchestrate that.

```
sudo chown root script.py
sudo chgrp root script.py
sudo chmod +s script.py
```

This will make script.py owned by root, and when anything runs it the 's' bit will ensure it will elevate the user automatically to root for you.

If it's a path problem, then you'll need to make sure that the .so file (the compiled shared object generated by the build process) is in a place that the python script that uses it can find it. This might require fiddling either with shell paths, LIBPATH or even the python path.

The best way to diagnose this would be to see if you can get a .lua or a .sh manually to shell into the python first (and fix any errors) before getting Domoticz to then initiate it.

Or you could in your .sh file capture the errors to a file?

```
#! /bin/bash
sudo python script.py 2> /tmp/errors
```

Then initiate that shell script from Domoticz and when it fails, look in the /tmp/errors file to see if there is any hint there as to why the script is failing to run inside the Domoticz context.

Lorccan, 2017-04-27:
I had already made the change to /dev/mem per the other ticket.

What solved it was a combination of an explicit path to the energenie module in the python script: `sys.path.append('/home/pi/energenie-master/src')`

and

`sudo chmod +s script.py`

I had never heard of the +s trick before.  I made the first change before reading your reply, so it may be that it's not absolutely necessary, but I am done tinkering with this switch for now.

I am now able to call the python scripts directly from the switch's ON|OFF Actions in Domoticz, so I am very happy.

Thank you David!

whaleygeek, 2017-04-27:
@Lorccan That's great news, well done for getting it working!

This might sound a bit cheeky, but I don't suppose you would mind posting a short article to the Domoticz forum, saying that you have your Energenie working with this code inside Domiticz now? I think if people can see that it is possible, we might attract more users to go that route, and we might get a nice critical mass of users that could help us to do further integration work perhaps?

Lorccan, 2017-04-27:
Absolutely.  You've helped me, so why wouldn't I!

whaleygeek, 2017-04-27:
Thanks @Lorccan - I've also reached out to Energenie and have let them know that we have some level of Domoticz compatibility now with this new code - this is great news, thanks for persevering and testing this!

whaleygeek, 2017-04-27:
@Lorccan I'll also let the Raspberry Pi Foundation know that we have some level of integration working now, I think they will be pleased, as Domoticz is well deployed on their platforms.

Lorccan, 2017-04-27:
One key thing that may not be immediately apparent is that the great advantage to the Energenie wall switches is that they don't need neutral at the switch.  Most other solutions do, and most homes are not wired that way...

Would you be able to have a quick look at the short piece of code that I am using for the ON|OFF please?  I don't know python and it may well be that some part of it is surplus (or plain wrong).  (I am not sure what led me, e.g., to `import domoticz`, but it doesn't stop it working!)

```
#!/usr/bin/env python3

import time
import sys
import domoticz
sys.path.append('/home/pi/pyenergenie-master/src')
import energenie

APP_DELAY = 1

energenie.init()
me_global = sys.modules[__name__]
energenie.registry.load_into(me_global)
ceiling.turn_on()
time.sleep(APP_DELAY)

# END
```

whaleygeek, 2017-04-27:
You might not need the APP_DELAY and time.sleep(APP_DELAY) in this example, because the turn_on() will run to completion, i.e. it will return once the radio has completely sent it's message.

It's probably not causing any problems, but it would be usual to have an energenie.finished() at the end of your script just to make sure the underlying GPIO pins are put into a known safe state before your script exits too.

If you already know the address of your ceiling light, you don't specifically have to use the registry, you could hard code it into the script, there is an example in the codebase that uses that form. However, I prefer to use the registry, because later on you can re-learn a different device and call it 'ceiling' and your script doesn't need changing. It's nice to have that level of separation between device details and the script that uses it.

It also means that if you have multiple scripts that do different things, you have all your house code and device codes in one neat place (registry.kvs).

So, I think what you have looks great.

Lorccan, 2017-04-27:
I think this is coincidental - but what a coincidence: I started to mod the ON script as suggested and got a failure (error 256) when I tried triggering via the ON action, so I reverted to the one that had worked and this failed with the same error.  The OFF script which has had no changes is doing the same.

I've tried restarting Domoticz, and also rebooting the RPi, but without success.

There's been no change in either system software or domoticz since this was working a matter of minutes ago, so I am mystified what could have happened.  Any thoughts?

whaleygeek, 2019-05-05:
Sorry I missed this one @Lorccan and it was a long time ago now. Note that some PR's have been merged recently that fix the sudo issue and also fix the pathing issue, both of which should improve the situation when running inside other applications like Domoticz.

I'll close this issue, but please re-open it if you have an outstanding problem that you need help with.

closed 2019-05-05 by whaleygeek