whaleygeek/pyenergenie #92: MIHO033 Open Sensor battery level...

RayMYP, 2017-11-22:
Hi,

In a few places I've seen suggestion that the MIHO033 open/contact sensor can also report battery level and that the MiHome gateway/app will show this. Unfortunately I don't have a gateway so can't prove or disprove it.

So I was wondering if you've left out the support in Devices.py for battery level because it doesn't exist or is there a chance you've not added support because you didn't know it was available?

Thanks

whaleygeek, 2017-11-22:
It's possible I missed it out, as I didn't have a spec of the devices when I was testing. However, the following code in the MIHO033 class indicates it should print an 'unknown' message on the console when any unhandled parameter is returned, and I don't remember seeing that.
```
            if "value" in rec:
                value = rec["value"]
                if paramid == OpenThings.PARAM_DOOR_SENSOR:
                    self.readings.switch_state = ((value == True) or (value != 0))
                else:
                    try:
                        param_name = OpenThings.param_info[paramid]['n'] # name
                    except:
                        param_name = "UNKNOWN_%s" % str(hex(paramid))
                    print("unwanted paramid: %s" % param_name)
```

RayMYP, 2017-11-22:
Thanks for the reply. I can't say I've seen that output but there's quite a lot of noise from the command line (lots of 'Can't decode payload' messages although all devices are operating as expected) and I'm loading it now in the background so not seeing the messages anymore anyway.

I'll see if I can log that somewhere else and monitor it for a while.

Do you know if the similar PIR devices always send the battery level or is it just occasionally or when low? My 033 sensors are fairly recent so not expecting them to go soon but would be nice to get the info out if it is there somewhere.

whaleygeek, 2017-11-22:
It's possible that the low battery status is only transmitted when the battery is low, rather than being a battery voltage. So the best way to test this would be to put a variable voltage power supply on the battery contacts and lower the voltage gradually, and see if you get any UNKNOWN messages come out with the battery status in.