whaleygeek/pyenergenie #93: Check more frequent : Energy Monitor

vanillabrand, 2017-11-26:
Hello - i am using the pyenergenie monitor with raspberry pi [314-RT radio board]. Is there any way to poll for energy stats from the device in more frequent intervals? I was hoping i could get to 5 seconds or less :) 

Am i right in thinking the Radio board simply listens to the monitor & the interval is hardwired?

Thank you in advance - great product & works well !

whaleygeek, 2017-12-02:
The timing of message sending is managed by the transmitting device - all that this library code does is to process incoming messages as they arrive.

I've checked through the documentation that Energenie tech team sent me, and there is nothing in there about instigating messages at a faster rate (either by sending a configuration message, or sending a poll message).

You could send an email to enquiries@energenie.co.uk and ask them if their product has other undocumented messages that we could implement in order to tease out data at a faster rate. If you find anything useful and public, could you post it back here so we can follow up on it? Thanks!

whaleygeek, 2017-12-02:
I have sent an email to Energenie querying if there is any more data about undocumented messages in the MIHO005 device firmware. However, I know they are quite busy and may take some time to respond.

The OpenThings protocol allows for a whole range of query and report messages, it just depends whether the device firmware supports it or not.

As a long shot, you could try forming a read message for the MIHO005 and FSK sending it to the device, to see if it triggers it to respond immediately. If that works, you could then poll on a 2 second interval yourself to get faster readings.

For example there is a MIHO005_REPORT template message here, 

https://github.com/whaleygeek/pyenergenie/blob/master/src/energenie/Devices.py#L139

That message already has the "wr" field set to false. I only used these templates for synthesising test messages, as per here:

https://github.com/whaleygeek/pyenergenie/blob/6f7d49d2788058ae8dd0bac6804765992694bbae/src/energenie/Registry_test.py#L161

So, if the device does respond to a "wr":False (which is essentially a query) and then respond by sending back these parameter values immediately (as OpenThings suggests it might do), that might work.

Here is an example of how an OpenThings turn_on() message is constructed from a message template, you could quite easily copy this out, replace it with a MIHO005_REPORT, perhaps don't put the values in (as this is a query for values) and then send it the same way, and see if anything comes back.

https://github.com/whaleygeek/pyenergenie/blob/master/src/energenie/Devices.py#L764

You need to get the radio back into receive mode as fast as possible though, as otherwise you'll miss the response. That is one of the reasons why in the current architecture the radio is left in receive mode most of the time, and only put into transmit when sending switch or join_ack messages. If you use one of the example programs, this should happen automatically at the low level driver (when you transmit it essentially pushes the receive state on a stack, changes to transmit, transmits a burst of messages then pops back to the receive state quickly in the C driver so that the delays in switching are minimised.

As in this use case the tx and rx are both in FSK mode, you shouldn't incur a radio reconfiguration delay (so avoid using any legacy OOK devices while testing, to keep the radio reconfiguration overheads to a minimum)

Let me know if you discover anything new.

whaleygeek, 2017-12-08:
I heard back from Energenie. The energy reports from this device come out at a fixed rate, and that cannot be changed and cannot be polled - this is because it performs integration of the readings over a fixed time period, and then transmits the message once all readings have been integrated.

So, unfortunately, there is no way to do this with an Energenie product.

closed 2017-12-08 by whaleygeek