whaleygeek/pyenergenie #94: Add support for new Energenie Thermostat MIHO069

whaleygeek, 2018-01-16:
https://energenie4u.co.uk/catalogue/product/MIHO069

From the product page:

The MiHome Thermostat is part of Energenie' s heating range of products and part of the MiHome home automation range. The heating range includes the radio controlled MiHome thermostat, and radio controlled Smart Radiator Valves, available separately. MiHome products link directly to the MiHome Gateway [available separately] which provides the communications between your MiHome device and the internet. The MiHome thermostat is suitable for switching any standard domestic boiler (see product specification below). The MiHome Thermostat connects to the boiler with a simple 2-wire connection and is battery powered. On-board controls to adjust temperature and turn the thermostat on/off are provided. The thermostat also senses occupancy and humidity which can be viewed in the App.

CODE: MIHO069

Achronite, 2019-03-24:
Hi - I know you have not provided support for this yet, but do you know the internal productId of this?  It is missed off in devices.py and I would like to support it in my node-red nodes.

whaleygeek, 2019-03-24:
Sorry, no idea. I am waiting for a loan device from Energenie so that I can power it up and see what comes back in the payload.

whaleygeek, 2019-05-15:
I now have a loan device. I plan to investigate this soon...

[image]
[image]
[image]

whaleygeek, 2019-05-15:
@jmigreen FYI

whaleygeek, 2020-01-13:
Yeah, sorry, just been super busy. Next opportunity I get I will plug it in and sniff the device id and update the devices.py