whaleygeek/pyenergenie #98: ener002- switching on/off

mstewart81, 2018-09-06:
I can get the sockets to connect to the raspberry pi and switch on/off after connecting but once the on/off test has been stopped what is the command to turn on/off

whaleygeek, 2018-09-06:
Hi @mstewart81 - the various programs are provided as examples of how to use the energenie library, so they are demonstrators only. The view is that you would modify these or write your own code based on them, to achieve the purpose you are building your system for. The specific actions that your system performs in order to trigger an 'on' and an 'off' of an unspecified number of devices, is specific to your application, so it's hard for me to provide a complete working example for such a case.

Code will have to be running continuously in order for the switches to be turned on or off repeatedly. So, if you use the setup_tool.py and leave it running, you can use the menu items to turn specific switches on and off whenever you want.

If you describe your specific application desire a bit more fully here, I might be able to steer you in a more determined direction or suggest which example code you should use as a starting point.

e.g.

What specific Energenie device types do you have connected to your system?

What are they controlling (tv, radio, light, etc etc)

Under what conditions do you want each of these to turn on or turn off (is it based on a timer, is it based on receiving an email, is it based on some other sensor attached to your Raspberry Pi, etc?)

whaleygeek, 2018-09-06:
There is a sample application built using this library of code in a recent MagPi magazine, written by PJEvans, that might be a useful example for you to read and modify?

https://www.scribd.com/document/380741586/MagPi70 page 18 and 19

Hope this helps

mstewart81, 2018-09-09:
I am looking at using 3 Energenie TRV but will build up for now could eventually be 6 in total but for now will concentrate on using between 1-3, I have 2 remote sockets to use with a phone charger and a light. 
I am reasonably new to RPI. but I have some experience with computers & programming cisco routers/switches

closed 2019-05-01 by whaleygeek