whaleygeek/pyenergenie #99: MIHO005 and ENER314-RT Pyton Code

frustratedgeek, 2018-10-13:
I've just bought a couple of MIHO005's and a ENER314-RT for my raspberry pi. I've got them all talking together using the demo code (each MIHO005 turns on and off in sequence).
But I'm struggling to write some code to manually control and read the data from each MIHO005. Has anyone got some insight?
I'm trying to write a basic webapp to control several appliances round the home. I've got everything all set and working, I just need some python to control and read these MIHO005's.

RayMYP, 2018-10-13:
Have a look in the registry.kvs file. It's just a text file and you can directly edit it if you're careful, back it up first to be safe. You should see the devices you've registered and they may have less than meaningful names, so as you can see I've renamed this example **swt_Power_EnergenieRF3**:
>    ADD swt_Power_EnergenieRF3
>    type=ENER002
>    device_id=[123456, 3]
> 

It's not the same MIHO005 device but I suspect the on/off features are the same. I basically only use the python library to integrate devices to my home automation server (OpenHAB) which is where I put all my business logic around if this do that and workflows etc. So my python code subscribes to an MQTT service for events like '/swt_Power_EnergenieRF3/ON' and it just does the action, also when it receives a message from ENER314-RT (via this great library) for a device it just publishes the event to the MQTT service.

But in its simplest form this snippet of python code would get the device from the registry and turn it on:
```
    ...
    energenie.init()
    # handy for debug and to show your registry
    show_registry()
    # ask the registry for the device and then manipulate it...
    device = energenie.registry.get("swt_Power_EnergenieRF3")
    device.turn_on()
    ...
```

frustratedgeek, 2018-10-13:
Thanks again Ray, Followed those instructions, created the registry file, and my 2 devices show up, all good.
In your original comment, you posted some code to get me started. Obviously if I just copy and paste that into a python file, it's not going to run by itself. I'd like to create a standalone python script with a simple function inside that will control one of the devices. I'm guessing I'd need to import something? All the help you can give would be really appreciated. Thanks.

jmigreen, 2019-04-28:
No need to create the registry manually. Let setup.py do it for you. 
For controlling it in a friendlier way, One can create a rest API for it using Bottle and its ability to both create a web server and automatically return dictionaries as JSON.  
If you add Cherrypy to it, you can do SSL too.

whaleygeek, 2019-04-28:
There is a mostly complete web console that I wrote using bottle on this branch...

https://github.com/whaleygeek/pyenergenie/tree/web_console

whaleygeek, 2019-05-05:
I think this issue is resolved now, so closing. If I got that wrong please re-open it, thanks!

closed 2019-05-05 by whaleygeek