whaleygeek/pyenergenie #100: Antenna rx gain too low - how to adjust 'pout'?

webash, 2019-02-27:
I sent back my mi|home hub in favour of getting the ENER314-RT module so I could roll my own energy monitoring with this library and [OpenEnergyMonitor](https://openenergymonitor.org) (emonCMS). However, I've found the RX range of the module to be underwhelming in comparison to the hub, despite the antenna and the rest of the architecture seemingly being the same.

When I contacted Energenie support, they came back with this:
```
The Antenna is the correct size for the frequencies used.  The ENER314-RT was intended for hobbyist use and it will be possible to optimise the antenna's gain by modifying the software.  

Pout range:  -1 to 13dBm in 1dB steps
```

Is it possible to do as they've stated while using the pyenergenie library?

whaleygeek, 2019-02-27:
Indeed. I will add a configuration option to the software to set power level for you, this is a great idea! A quick hack if you are in a rush would be to change one of the initialisation table entries in drv/radio.c and re run the build script. I will look up the datasheet later and point you to which entry to change.

whaleygeek, 2019-02-27:
Register 0x11 sets Pout...

[image]

[image]

However, a value is never written to this register by the config table, so it should be at default, which looks to be already at the highest Pout. It is *possible* that this default is too high and the (enabled) current trip is firing. So you could try lowering the power level slightly (I know that sounds counter intuitive).

It will be a while before I have the time to reconstruct my test setup and measure the end to end system effects of this with a power meter, so don't wait for me.

Add 0x11 for address, and follow the PA configuration suggestions in the attached screen shot to select a non-default value, and see if it improves your range.

Add the address/value to both config tables here:

https://github.com/whaleygeek/pyenergenie/blob/master/src/energenie/drv/radio.c#L124

Then while cd'd into the drv folder, run

```./build```

To rebuild the .so library, and re-run the Python.

Radio power is affected by inverse square law...

https://www.rfvenue.com/blog/2014/12/15/when-it-comes-to-rf-distance-plays-tricks-with-the-mind

```[quote] Think of it this way: double the distance, and you get four times less power. This works the other way around, too, and is crucial to the point we're trying to get across here: halve distance, and received power is increased four times over. So, while you can't always have the freedom to move a receiver as close as you want, shortening the distance by just a few inches can counterintuitively produce dramatically better signal, because of the inverse-square relationship...```

You could run a repeat loop that turns your switch on and off repeatedly, and measure the edge of the reception range line of sight - change the register, re-build and re-run and see if the reception range changes.

Hope this helps.

webash, 2019-02-28:
Hey! Thanks for this! Super detailed and helpful in comparison to energenie support! I'll give this a go over the weekend. At the moment I seem to have jimmied a workaround by wedging a key into the antenna! Haha! It can now successfully see the Whole House Monitor in the meter cupboard.

However, I do need that key so I will be trying this soon.

closed 2019-05-01 by whaleygeek

webash, 2019-06-04:
I never got around to trying this. My literal key hack is still in place. I'll come back and provide an update when I finally do, as I'd still like the key not to be necessary...

whaleygeek, 2019-06-12:
@webash I spoke to Energenie about this a few weeks ago. 

The issue is probably RF related, they said (and I agree) that grounding, antenna positioning and other physical effects of construction will have a negative impact when using a plug on board for the Raspberry Pi. 

For their MiHome hub, they would have been able to have full control over all system design parameters, but with an attachment board for a (third party designed) Raspberry Pi, they don't have full control over the whole system, so cannot optimise it as a system.

Your key hack sounds a good one actually, I wonder whether you have effectively doubled the antenna length as a result, so it's a half-wave rather than quarter-wave??!!

webash, 2019-06-12:
Yeah definitely understand the challenges of a radio attached to unknown other hardware. I've tried the antenna in all sorts of angles, positions and so on; but the only reliable method was this 'key hack'. As I'm using the Raspberry Pi 3B+ with the official AC adapter, I would've expected it to be as close to 'standard' as possible. Perhaps it is worth trying other adapters; otherwise nothing else is connected to the board (except the official case, with the lid off).

You might be quite right about the antenna length! When I contacted support about this they weren't super helpful, other than providing this information about changing the `pout`. I do still intend to experiment per the help you gave above, it just hasn't been convenient to dig into that level.

webash, 2020-02-07:
I haven't yet tested whether I can remove the key, but I was getting further interruptions on communication from the whole home energy monitor... I then remembered something about RF interference on the Raspberry Pi when using WiFi... so I disabled WiFi and Bluetooth using the boot configuration options... and boom... no more intermittent signal to the whole home unit...  so perhaps that has been the problem all along. Will let you know whether I can now remove the key entirely :)

Achronite, 2020-04-29:
> 
> 
>  Will let you know whether I can now remove the key entirely :)
Just to let you know, I also had issues with range on the ENER314-RT, which I have fixed using the same method I used on my ENER314 board, - by soldering a 173mm piece of straight wire to the existing coil antenna. where it is attached to the electronics on the board; it works really well.

webash, 2020-04-30:
@Achronite this is amazing knowledge! any further details about the gauge of the (copper?) wire? I don't currently have a soldering iron, but this might be a good excuse to finally do so, because it would be nice to not have my range impacted every time I need to check the power and water meters 😄

Achronite, 2020-05-02:
[image]
I just used a solid core interconnect wire (0.5mm-ish), the key is to make it exactly 173mm long to match a quarter wavelength.  Having it straight also helps.

webash, 2020-05-03:
Amazing! Thanks very much!

webash, 2020-05-08:
Done! Still losing about 1 in every 10 or so packets, but there are a lot of thick walls and building risers between the whole home monitor and the Raspberry Pi; so I'm more than happy that this is better than having a key wedged in it!

Achronite, 2020-05-10:
It's more likely that it's radio interference causing the dropout.  Loads of stuff operates at that frequency. I built in auto-retry into my node-red plugin to get round this.

webash, 2020-05-18:
If I hadn't sunk in a bunch of time into (somewhat unnecessarily, but it was fun) building a wrapper around pyenergenie to push to MQTT, then I'd probably switch to your Node Red plugin; thanks for keeping watch/suggesting things back in this library.

genestealer, 2023-11-26:
> [image] I just used a solid core interconnect wire (0.5mm-ish), the key is to make it exactly 173mm long to match a quarter wavelength. I suppose copper would be best, I'm not entirely sure what my wire is. Having it straight up also helps.

@Achronite You need to cut the original loop (or de-solder it).
[image]

Achronite, 2023-11-27:
> @Achronite You need to cut the original loop (or de-solder it).

@genestealer It didn't seem to matter for me, I got the signal boost I needed, and leaving it connected seemed less destructive to me.

whaleygeek, 2023-11-27:
The Pout is chosen to fit within the spectrum power mask requirements for that challel, but Pout is not directly related to ERP (total system power) as there are many other factors. The MiHome hub has a proper groundplane design and the radio placement is such that ERP is maximised for the given Pout. On the Raspberry Pi with the RT board attached, the RF performance will be worse, because grounding and RF interference from the CPU would not have been optimised as a whole system.

genestealer, 2023-11-27:
Absolutely agree about interference from the CPU. I'm currently using a free samtec ribbon cable sample to extend the pinheader, so the transmitter board is outside of the metal Pi enclosure I'm using.

I've also ordered an antenna with cable which I will solder the wire directly to the board in place of the existing coil antenna. https://a.aliexpress.com/_EGOLbrf