whaleygeek/pyenergenie #102: The future of this library and the Energenie clone

webash, 2019-03-04:
Not sure an issue is the right place for this discussion, but in lieu of chat or other discussion forums on GitHub, hey what the heck.

Originally I commented on that PR in the Energenie repo *before* I came across your far more fresh repo here. I've since been pretty happy with this repo and it's progress, not to mention your responsiveness. 

However, you alluded to a future of the Energenie devices/libraries that may be worth discussing. Can you elaborate @whaleygeek?

whaleygeek, 2019-03-05:
I have an open invite to visit Energenie soon and chat to them about them perhaps pulling all my changes into their repo. Both sides have been busy with other stuff so I've not had a chance to organise this yet, but will try soon.

webash, 2019-03-05:
That's great to hear. Thanks for your commitment to this, I would've been stumped on my home energy monitoring project without your work here.

iotic-labs-markwharton, 2019-03-06:
Top work as usual, @whaleygeek . It indirectly won our first funding round when we "magically" turned on a toaster in our investor's board room.

closed 2019-05-01 by whaleygeek