whaleygeek/pyenergenie #103: Modified __init__.py to support running from any location

jmigreen, 2019-03-09:
I noticed that my code would only run when the current working directory was inside the pyenergenie folder. After some sluthing, I found that the code in __init__.py was hardcoded to only use the default file name of the registry without a path. 

I have modified it to join that filename to the sys.path[0]  path. This allows derivative scripts to be run from any location including rc.local. Apologies in advance if there is a more correct way of doing this on github. I'm an amateur... 

The modification begins at line 48. 

` path = os.path.join(sys.path[0], registry.DEFAULT_FILENAME)`
   ` if os.path.isfile(path):`
       ` registry.load_from(path)`
        `print("loaded registry from file")`
        `registry.list()`
       ` fsk_router.list()`

whaleygeek, 2019-03-09:
Thanks, this is a good fix. I tend to do this now in other projects, but have not applied this logic here yet.

I see you joined github recently, so I won't suggest you send this as a PR. I will leave this issue open, and next time I make changes to the library I will include your fix.

I tend not to make changes that often, as I have to test everything again and that takes time, and most users of this repo prefer stability over constant change. I will fix it next time I do some work on this though.

Thanks for your contribution, and welcome to github!

whaleygeek, 2019-03-09:
Fixed by applying this PR https://github.com/whaleygeek/pyenergenie/pull/104

closed 2019-03-09 by whaleygeek

jmigreen, 2019-03-15:
Thanks for such a warm welcome :) I've been having fun learning the ropes.