whaleygeek/pyenergenie #109: Rest API for setup tool?

jmigreen, 2019-04-28:
Hi David, 
I've built a Bottle-based rest api around the python code controlling my MiHO005s at home, and I've written a little iOS App to interface with it. Working great!

Only thing I haven't quite wrapped my amateur brain around is how one might go about doing the same for the setup tool so that discovery can be done. So many of the scripts in the library seem to assume cmdline I/O.  

Any thoughts or suggestions on where to start?

whaleygeek, 2019-04-28:
Hi!

There is a mostly complete web console on this branch, that includes a legacy_learn page, and a data logging page. You could use that as a basis for driving the setup functionality. The discovery behaviours are configurable so you can re code them to work in a web context.

This also uses bottle.

https://github.com/whaleygeek/pyenergenie/tree/web_console

jmigreen, 2019-04-29:
That's pretty cool! 
Yeah,  that should jumpstart things quite a bit! Thanks!
Since it's going to an iOS app, I'm tempted to try and add something like OAuth to identify the device, but  I have no idea whether that's simple or a nightmare though in python.

whaleygeek, 2019-04-29:
Thanks!

Anecdotal evidence on OAuth is mixed. The twitter API was relatively easy but there were python libraries specifically for twitter. I have heard others say that OAuth takes a bit of time to get working, but there are libraries for it...

https://pypi.org/project/oauthlib/

Please feel free to nick things from the webconsole branch. I might finish it off one day. It does work, but I hadn't formally tested it with all devices, which is why I was holding back merging it to master.

whaleygeek, 2019-05-05:
This might be a good one (rest api for setup) to add to the work that is going on in the webconsole branch. I'll have a think.

jmigreen, 2019-05-05:
That would be cool. Bottle makes it very easy to return JSON. If you return a nested dictionary from function it called, it automatically sends it back as JSON. So the initial setup for my app was surprisingly simple. 
The web console branch helps save me from having to re-work the setup functions too.
I haven’t had a chance to get started on it yet, but it’s definitely on my to-do list.

-Josh

whaleygeek, 2019-05-06:
Oh yes, good point, the JSON rendering bit is pretty much automatic!

jmigreen, 2019-08-12:
Hi David,  hope you're well. 
After a bit of a hiatus, I've gotten my api somewhat harmonised with the web_console code so that I can get a modal browser to popup in iOS when I want to scan for/add new devices. 
I've got one minor blocker that I can't quite figure out how to resolve:

When I quit the server, I get the following warning:

`ResourceWarning: unclosed <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0`

I've checked in netstat, and it does appear that the socket is being unbound, but I'm guessing that something about the new session logic is not being cleanly closed. Any thoughts?

whaleygeek, 2019-08-12:
Hi - all good thanks! 

Hard to say really without looking at your code - is it on github? I can't find it in your repo list?

What state is the socket in, in the netstat listing? Is it in TIME_WAIT?
Which end closed the socket (did any end close the socket and are you sure it did - put a print just before the close() call)

You can sometimes change socket options to improve the situation...

https://notes.shichao.io/unp/ch7/

Look at SO_LINGER.

jmigreen, 2019-08-19:
Hi David,
I've added you as a collaborator on the repo. I should ideally fork the
web_console branch and upload the work there, but I was lazy when I first
downloaded it, and creating a separate repo seemed safer?
It's still very much a WIP, and undoubtedly very amateurish,  but both the
web console side of things and the switch controls seem to work! I'm just
curious about the socket warning I keep getting as I. have no clue
precisely where it's coming from or how to fix it. Any advice would be
hugely appreciated.

Kind regards,
Josh

whaleygeek, 2019-08-22:
I had a quick look at this.

It really does depend how you are stopping the web server running. A CTRL-C will likely generate a resource warning (anecdotal evidence here too.... https://github.com/psf/requests/issues/3912) and the PSF guys say 'this is only a warning, don't worry about it').

There is some advice here about how to access the shutdown() method, as I suspect because the web_console is session based (and I use cookies in session to implement the session) that Bottle is using the Requests library internally, which will suffer on an uncontrolled shutdown.

https://stackoverflow.com/questions/11282218/bottle-web-framework-how-to-stop

I'll leave these notes here, next time I do some work on web_console I'll see if I can add a'controlled server shutdown' feature that you can call from a web request (which would allow you to add a button in an admin interface to stop the server remotely, if that is what you want to achieve).

But as the PSF guys say, it's just a warning, it's sort of normal when you pull the plug on a server that is actively serving.

whaleygeek, 2019-08-22:
FYI... https://github.com/whaleygeek/pyenergenie/issues/118

jmigreen, 2019-09-16:
Overlooked one thing though: mobile-friendliness. It looks like if I converted it to use Flask instead of Bottle there are some plugins for switching between desktop and mobile templates automatically. Am I missing any more obvious/easy alternative?

whaleygeek, 2019-09-16:
Bottle is definitely a tiny Flask, I'm not an expert in either. My reason for using Bottle was it was small and simple, could easily be embedded in this repo, and all I really wanted to achieve was to provide a very minimal setup tool to show how it could be done. A Flask based implementation would be bigger but more feature rich, and something that would be worth exploring for your specific needs. I've seen some nice, responsive, portable websites built with Flask.

I don't think the web offering in this repo would ever grow beyond bottle, as it is only meant to be a demo of it 'being possible to interface with Energenie devices via a web interface. But I would watch your progress with Flash with interest, as it may inform things that could be improved in the lower level interface (threading, for example) to make it easier to integrate with other frameworks.