whaleygeek/pyenergenie #116: Support for MIHO109 Wifi Plug

whaleygeek, 2019-05-15:
I have one of these devices on loan to evaluate, the new WiFi plug. It would need to be controlled from the portal, but that is free and documented here: https://mihome4u.co.uk/docs/api-documentation and specified using MoonRope: https://github.com/adamcooke/moonrope

I wanted to test people's appetite for using this plug with the Raspberry Pi and this library- it's very small and neat, about a 10m WiFi range, but would require me to implement an API connector for the MiHome portal (which looks like it could enable other goodies like iPhone and Android integration, geofencing, remote access, etc).

[image]

whaleygeek, 2019-05-15:
@jmigreen FYI

jmigreen, 2019-05-15:
So this uses an Energenie-hosted API? So the RPi would just be replacing the role of the hub their? Would be interesting in terms of gaining the geofencing, IFTTT integration stuff. I guess that means their own iOS app would work too. Certainly a huge convenience for those not keen on building their own iOS app!

whaleygeek, 2019-05-15:
Yes, that is correct.

It is the only way to get the new WiFi plugs integrated into the system, which is what started me thinking about this line of attack.

webash, 2019-06-11:
I guess the cost/reward ratio would need to be weighed up. What else could be gained by interacting with the Energenie APIs from the library?

whaleygeek, 2019-06-12:
Direct access to the MiHome web interface and mobile apps, and possibly better range with wifi than the 433 radio. However there is currently only one Energenie product that uses Wifi, so the breadth of available devices will be worse. But I've seen the WiFi plug working and it's pretty neat.

ruberyguy, 2020-09-15:
Just seen this thread so appologies if this is past the point of interest. However, I saw this device for less than a £10 so I bought one thinking along the same lines as you. It seems to have a simple http interface so quite simple to control. There is one thing concerning me though. Why do I need to go via a an external web site. Is this correct? What do they gain from this other than they can see my requests and they have access to my home network. I should be able to talk to the device directly and not need an account. Has anyone tried this? The issue is to give it an IP address you need to give the server your wifi password. Hopefully I'm missing somthing here as this would be a major securty issue. Of course I could just trust them.......

As for the WIFI range. Surley this is not so good as 433. I have low powered 433 devices that work half way down my garden 50M. Wifi, on the other hand, can't reach my back room. Ususally the higher the frequency the worse the range. I've heard that 433 devices on low power can achive more than 100M and as most messages are simply off and on the bandwithdth is not an issue.

whaleygeek, 2020-09-15:
Hi @ruberyguy 

Going via the external website means you can turn things on and off when you are not in close proximity, so you can turn your living room lamp on and off while at the shop, if you wanted to. This would not be possible otherwise unless you had a static IP, VPN, or other complex networking setup.

The device itself supports wireless host mode and wireless peripheral mode. It only seems to use the wireless host mode for pairing, presumably so you get the cloud remote control benefits via your ISP. Also, you can set up timers and other things via the cloud interface, and link to IFTTT for event driven operations, that would not be possible if you used it in wireless host mode and spoke to it locally all the time. Also, a mobile phone can only have a live association with one base station at any time, so if you had 4 devices in your house all in base station mode, you would have to disconnect from one and connect to the other to control it. I'm not aware of it supporting ad-hoc network mode, but that would limit you to 3 devices (and no internet!) anyway.

Giving it your wifi base station id is no more insecure than doing same to your smart TV. If you are like me, you use the guest network on your wireless router configured to *only* give internet access and no LAN access, to limit the scope of any damage. Beyond that, you always have to trust device manufacturers that their devices have secured the password, encrypted or otherwise.

Regarding range etc. 433MHz is in the ISM (industrial scientific and medical) band and it suffers huge interference, more so than the 833MHz band (and a lot of consumer devices are now moving over to 868). There is no protocol for spectrum sharing in that band (unlike WiFi), so messages either get through or they do not. Usually (the Energenie and my doorbell and PIR sensor all alike) just send 10 message repeats in the hope at least one gets through, and there is no end to end acknowledgement (unlike WiFi) so you cannot confirm receipt. There is also no spectrum sharing protocol on these devices (no collision avoidance/detection/recovery) so the usage of the 433/868 spectrum is generally quite poor.

WiFi of course is standards compliant, high bandwidth, end to end acknowledged, built in security, time and frequency division sharing (so better use of the spectrum). If you are suffering WiFi range issues those are more likely because everyone else in your neighborhood has high power WiFi which is generally heavily used (an on/off doorbell or switch might send one or two messages a day). So WiFI being quite long range now means that actually you are sharing the limited spectrum in that space division multiplexed zone with lots more high power devices, and as such there is generally less available bandwidth available. I wonder how many wireless 433 doorbells pressed twice per day there are in the local area compared to high power MIMO (multi channel) range-max WiFi routers constantly streaming NetFlix!

Hope this helps.

ruberyguy, 2020-09-15:
Thanks for fast reply, yes it is helpful. I do worry about the number of devices inside my firewall. For these devices to be secure I would use a guest WIFI account so the device cannot see my network. However, then I can't see the address either locally although perhaps there is a way as I use openWRT. I will give it a go.

I speak to the devices via my Raspberry PI so anything is possible :-) and my phone talks to the my PI.

The advantage of ISM is that it would be difficult to use this method to attack my network so I feel quite safe using devices.

Interesting info about interference though. My PI and the 8 eTRV don't miss a message. I have also converted an Openenergy an emonTH device (Remote temperature) to talk to my PI and again no issues with missed message. 1 or 2 a month. However, I've just added MIH005 (adaptor plus) and this is proving very unreliable. But not sure this is interference as temporarily removing power always brings it back immediatly. Very odd. I'll do some more testing but I'm not sure this is quite straight forward.

whaleygeek, 2020-09-15:
No problem.

Just one point, hacking into ISM is really easy, especially with either a SDR dongle or even the energenie radio. A capture replay attack is easy on ISM if you are in range - just use HackRFOne and GNURadio to capture the payload, then play it back.

Indeed, that is easily possible with the HopeRFM69 radio chip used on the energenie two way radio board.

WiFi is much more secure, mostly because there is a protocol behind it, and mostly because ISM devices just respond to the same identical pattern every time.

whaleygeek, 2020-09-15:
I would like to add a diagnostics mode to this library at some point. It does sort of already display malformed FSK messages if it receives them. The Adaptor Plus is FSK and bi-directional, whereas the on off switches are OOK and unidirectional. You'll probably get different performance with the two different modulation schemes, as they will handle disturbances differently.

Someone on here stuffed a house key inside the aerial to get a longer range, I suspect on the Pi it's mostly about ground planes and harmonic interference from the processor and other signals that affects sensitivity.

ruberyguy, 2020-09-16:
I realise ISM is easy to hack but the rewards for the hackers are trival in my case. They couldn't gain access to the PI or my network. Hackers would also have to be local and if they decide to switch my radiators on so be it. What I don't understand when manufacturers offer services that are free I always think, so what's in it for them. So I'll choose ISM for the time being. Don't mind if its hacked and prefer the range. I would only allow the WIFI devices access to the internet and not my local network for peace of mind.

Still think there is an issue with MIH005  which is not related to noise. When plugged in, if it initially doesn't respond after a few attempts it doesn't seem to respond at all. Remove power and start again and it seems to always start working. Difficult to prove but moving the item closer should improve it which it doesn't. It seems to behave the same no matter what distance it is from the pi. However outbound messages from the adaptor always works. However, when its working it does seem to have short periods where it doesn't respond to on/off requests. Diagnostics will always be useful although I think the issue is on the MIH005.

As you say as the MIH005 is two way so at least I can check the state and resend request. As this item is less than £20 it is one of the cheaper ways to control a device so is my prefered method at the moment.