whaleygeek/pyenergenie #117: Add RSSI Measurement API

whaleygeek, 2019-05-22:
I would like to read RSSI measurements for received payloads, to help better characterise the performance of the Pi RT adaptor range. 

This should ideally read out the RSSI value of a received packet, and store it in a field somewhere in the device class so that the RSSI of the last received message for each device is available. (RSSI will likely vary for each device due to different distances, hardware performance of end devices and obstacles. The time of the last received message will also help place the RSSI measurement in a temporal plane.

whaleygeek, 2019-05-22:
https://github.com/whaleygeek/pyenergenie/issues/66

webash, 2019-06-11:
Adjusting the power level per #100 seems like a related issue/feature request

whaleygeek, 2019-06-12:
Yes, thanks for linking it up, that was indeed what I had in mind here. If we can estimate RSSI as received by the Raspberry Pi, we may be able to get a better idea of range by taking multiple measurements at different distances. 

RSSI at the other end might not be the same, of course, as the performance will vary for many reasons. But it could be a good indicator to help people decide where to site their Raspberry Pi for best effect. I'm thinking of a 'signal strength surveying' tool as part of setup_tool.py might be a useful addition.

webash, 2019-06-12:
Sounds great