whaleygeek/pyenergenie #119: MainLine Driver not working on Raspberry Pi 4

asison-witekio, 2019-10-15:
This issue is specifically directed at the mainline repo.(https://github.com/Energenie/pyenergenie)
I use monitor.py to check for energenie devices. This always worked when I use Raspberry Pi 3.
I got a Raspberry Pi 4 and inserted an SD card containing my pi3 image. I ran monitor.py on RPi 4 and got this:

```
warning: method is untested:<function receive_len at 0xb65bac48>
monitor:starting monitor tester
```

It got stuck there for 1 hour with no response.

I did  some further investigation. I enabled logging on the "energenie/drv/" (c driver land).
I compiled it and ran monitor.py on both RPi versions and got the following results:

```
####### RPI 3
pi@rpilab0003:~/lab/energenie/pyenergenie/src $ sudo python3 monitor.py
warning: method is untested:<function receive_len at 0x765bebb8>
monitor:starting monitor tester
radio_init
reset...
reading radiover...
36
radio_standby
writereg 1 4
_wait_ready
radio_receiver
writereg 2 0
writereg 5 1
writereg 6 236
writereg 7 108
writereg 8 147
writereg 9 51
writereg 11 0
writereg 24 8
writereg 25 67
writereg 3 26
writereg 4 11
writereg 46 136
writereg 47 45
writereg 48 212
writereg 55 160
writereg 56 66
writereg 57 6
writereg 1 16
_wait_ready
monitor:receiving payload
```

```
####### RPI 4
pi@raspberrypi:~/lab/energenie/pyenergenie/src $ sudo python3 monitor.py
warning: method is untested:<function receive_len at 0xb65a8c48>
monitor:starting monitor tester
radio_init
reset...
reading radiover...
0
warning:unexpected radio ver<min
radio_standby
writereg 1 4
_wait_ready
```

Comparing the 2 monitor.py logs, below "reading radiover...", RPi 3 reported 36 which is
the expected radio version extracted from the radio module through SPI (bit banged). On the other hand, RPi 4 reported 0.

"_wait_ready" is when the driver waits for the radio module to give a response. On RPi 3, it gets through it. On the other hand, RPi 4 gets stuck on the very first "_wait_ready" which means that it's not getting any messages from the radio module.

On my comparison, there is definitely an issue with sending/receiving SPI data
between the radio module and Raspberry Pi 4 (specifically).

I tested this repo(waleygeek) and found that you need to rebuild the driver to make it work. So this repo works on Pi 3 and 4 (ran mihome_energy_monitor.py). The problem with this is that we have an application that relies on the older library and we would have to rewrite a good portion in order to interface with the waleygeeks library.

I tried using the rebuilt waleygeek *.so file and placed it in the mainline repo’s “/drv” folder. To my surprise it worked.

CONCLUSION:
We need the mainline repo to support Raspberry Pi 4 ASAP or we will have to design our system without the product

NOTE: Someone else already bumped into the same issue. (https://www.raspberrypi.org/forums/viewtopic.php?t=246495)

whaleygeek, 2019-10-15:
I will check into this, but I believe there was a community PR that added (yet again) new address details for the GPIO block for the new BCM chip used on RPI4 - it is possible that the community PR did not rebuild the .so file that is cached in the repo when doing this. I will check first to see if this is the issue, before raising it with Energenie to fix their repo (as I don't manage that one)

whaleygeek, 2019-10-15:
I think this was the PR, it was to do with 'not running as root', I think permissions changed for GPIO across OS versions (rather than it being a board variant). I note that the cached .so file was not rebuilt by this PR, so the out of the box .so will be old if using the latest master. I will definitely fix that part.

https://github.com/whaleygeek/pyenergenie/commit/4bc3f7d529428ecc9daa92cd7d653ec1757a3537

We may have to send a small patch to energenie for the C code without changing all the Python, so that it doesn't break your code. 

Do you have a link to your product that we can look at?

Also, are you trying to get users to be directed to the Energenie repo to get code? Would directing them to your own repo with a fork of the fixed code suffice? Long term plans are to get Energenie to update their repo to keep it in step with mine, as their code is really out of date and the new Python abstractions I added since are much easier for users to use.

David,

philprobinson84, 2019-10-15:
Hi @whaleygeek, thanks for the quick response!

I manage the UK business unit @asison-witekio works for within Witekio.

We have an internal tool we call the Automation Lab (we're planning to open-source this tool in 2020) - this is used for high-level development and regression testing. One of the power distribution units we support is an ENER314-RT radio and the MIHO smart sockets. We've got multiple instances in the field running great with the Pi3, but ran into the problem described above when building a further 8 instances of the lab using the Pi4.

Thanks for the suggestion regarding forking from your repo - I think this is likely to be the route we head down.

asison-witekio, 2019-10-15:
Hi @whaleygeek , thank you for the quick response. 
Just a side note, I have been running all the the scripts as admin on pi 3 and 4.

Achronite, 2020-01-13:
@whaleygeek the code just needs recompiling and updating in the repo to close this.  I didn't include the .so in my original pull request, the one in src/energenie/drv is 4 years old.

whaleygeek, 2020-01-13:
Noted. Sorry, been busy with other projects. Will have to dig out my Pi and rebuild it.

whaleygeek, 2021-04-14:
Fixed by https://github.com/whaleygeek/pyenergenie/commit/9a7abe11b748b50e970a5a26bae82e4448a7892f

closed 2021-04-14 by whaleygeek