whaleygeek/pyenergenie #121: Remove all exit calls

whaleygeek, 2020-02-06:
See: https://github.com/Achronite/energenie-ener314rt/issues/8

@Achronite thanks for raising this, added it to my list.

whaleygeek, 2021-04-14:
ANALYSIS:

This appears in 2 places in gpio_rpi.c, both in gpio_init()

gpio_rpi.c: 85         exit(-1); //TODO return a result code
https://github.com/whaleygeek/pyenergenie/blob/master/src/energenie/drv/gpio_rpi.c#L85

gpio_rpi.c: 104     exit(-1); //TODO return a result code
https://github.com/whaleygeek/pyenergenie/blob/master/src/energenie/drv/gpio_rpi.c#L104

The gpio_init() function should be changed to return a GPIO_RESULT that is 0 if ok, and non zero on fail. This can then be checked in the caller (and presumably propagated out in the same way by spi_init() using SPI_RESULT, then radio_init() can return a RADIO_RESULT.

radio.py init() must then check the return result, and raise an exception presumably if it is non zero.

This will then cause __init__.py init() to throw the exception, then calls to energenie.init() will propagate the exception into the main program. The main programs in all demos should then catch this exception and fail with a message and then hit a finally block, that tries to clean up the GPIO state so it is in a sensible default OFF state (thus allowing another run of the code to succeed if the hardware problem is now fixed);

Achronite, 2021-04-16:
My fix can be seen here: https://github.com/Achronite/energenie-ener314rt/blob/master/C/energenie/gpio_rpi.c#L57