whaleygeek/pyenergenie #123: MIHO009 Double Light Switch

genestealer, 2020-08-13:
Anyone know of any reason why the MIHO009/MIHO071/etc **Double** Light Switches are not working with this?
Are they operating on a different encoding?
The manual for the Double Light Switches https://energenie4u.co.uk/res/pdfs/MIHO009-User-Guide-v1.0.pdf seems to be missing infomation on the radio when compaired to the Single Light Switches https://energenie4u.co.uk/res/pdfs/MIHome-Light-User-Guide-v1.5.pdf

whaleygeek, 2020-08-13:
I don't have one of these devices to test against, but...

The tell tale statement in the doc is this:

• Radio: 433.92MHz ISM short range receive only

This usually means it uses the OOK protocol, and the learning is done on the switch itself. I'm not sure what role the two buttons play in pairing - it is possible you pair them individually, although I can't ascertain if that is the case from the user guide. If you have a hand-remote, you could check if you can independently pair the left and right buttons to different buttons on the hand remote.

Original Mihome OOK devices used to have 2 memory locations for pairing, looks like they may have upgraded this to 3.

Note that the range on the Raspberry Pi radio is significantly worse than the official mihome gateway, due to the various grounding, casework and aerial differences that are out of control of Energenie at the Pi end.

Also, putting the light switch inside a metal earthed box(!) will have an effect on radio range, and if the Pi is geographically behind the switch instead of in front of it, you effectively have an earthed cage in the way, and only a reflected signal will make it into the switch (presumably via the front plastic plate).

I suspect it is more a reduced radio performance issue than anything else.

Look at the aerial hack picture in this ticket on May 2nd, and try that and see if it helps.

https://github.com/whaleygeek/pyenergenie/issues/100

Also, you don't say if you have other devices that DO work with this library. Do you have a faulty radio?

genestealer, 2020-08-13:
Yes the two buttons each have their own independent program memory. Programming is meant to be the same as the single button, press, hold, led flashes to say it's in program mode, but it then ignores any of the TX On-Off signals, even the ones which I know work with the single lights (they are turning off and on when im programming these new one, so I know the TX is working ok) 

Yes, it's just this double switch I'm having issues with, the singles are working fine. I will check about the reduced radio performance, as I do have a metal earthed box.

genestealer, 2020-08-14:
So I tired with the light switch front cover off, at about 50cm distance: still nothing.
I've also not checked the equivalent single light switch with slave (I have the double with slave), https://energenie4u.co.uk/res/pdfs/MIHO43-User-guide%20WEBSITE%20COPY.pdf which does include text regarding pairing the handheld remote, which is missing from the double light switch. Very odd.

genestealer, 2020-12-19:
It turns out I had a faulty night switch and replacement double light switch works just fine.

closed 2020-12-19 by genestealer