whaleygeek/pyenergenie #125: Device support matrix

setdetnet, 2020-11-16:
Excellent effort with this.
What would be helpful is a matrix of device (OEM codes like <MIHO003, MIHO013 and so) on  down the left, and across the right - the current support level (fully, partial, future) and perhaps a comment. There's quite a few devices. I couldnt find the status for MIHOI013/14 that I have - happy to help if you ask me what to do with them (sniff etc) and how.

Achronite, 2020-11-16:
I've done a similar matrix for my node-red and node.js code, which uses this library see [https://github.com/Achronite/node-red-contrib-energenie-ener314rt](https://github.com/Achronite/node-red-contrib-energenie-ener314rt#supported-devices).  The eTRV (MIHO013) was a pig to get working, due to it's small receive window.

I would be very happy to see traces from other devices that have not been tested yet.

whaleygeek, 2020-11-16:
Good idea.

As a general rule for now - any device that is receive only, uses OOK modulation and the simple bitwise encoder, so there is no device ID in those, they just learn whatever the transmitter sends out (and by design are therefore compatible). A lot of the sockets and all the new light switches are receive only, so they are supported as-is.

The only device that I know is not supported at present is the thermostat, but I have one of those here and all the data necessary to get it working.

whaleygeek, 2020-11-16:
Regarding the eTRV, I have an outstanding issue to make the transmit and receive a scheduled activity so it will then be easy to achieve the tight receive window. I'm just waiting for time and motivation to do this!

whaleygeek, 2020-11-16:
There is a list of devices, their product names and their device id's in the Devices.py file here: https://github.com/whaleygeek/pyenergenie/blob/master/src/energenie/Devices.py#L24

If someone was to paste a list of other product numbers that are not listed (paste it in this issue) I will bring it up to date - I have some extra data here that Energenie sent me also, so I can consolidate it all and update the existing list.