whaleygeek/pyenergenie #131: MockSwitch failure

whaleygeek, 2021-07-28:
Reported via Energenie support:

build_rpi runs successfully and setup_tool runs successfully, but
I still get the error below from mihome_energy_monitor.py.

The only thing that I don't do is use the Raspbian Buster OS (I
use ARM), but I don't have the impression that the OS is the
problem here, because I have installed all of the dependencies
successfully and receive no error messages relating to libraries.

Are you able to provide any information about why this error
might be occurring so that we can try to identify the source of the
problem?

Starting energy monitor example
loaded registry from file
REGISTERED DEVICES:
mock -> MockSwitch(0000)
2gleft -> MIHO009(0x123456,0x1)
2gright -> MIHO009(0x123456,0x2)
plus -> MIHO005(0x373)
dimmer -> MIHO010(0x6c6c6,0x0)
auto_0x1_0xa56 -> MIHO004(0xa56)
ROUTES:
Logging to file:energenie.csv
Traceback (most recent call last):
File "/home/alarm/local/src/pyenergenie/src/mihome_energy_monitor.py", line 64, in <module>
energy_monitor_loop()
File "/home/alarm/local/src/pyenergenie/src/mihome_energy_monitor.py", line 33, in energy_monitor_loop
d.set_switch(switch_state)
AttributeError: 'MockSwitch' object has no attribute 'set_switch'

For information, when running setup_tool, I receive messages:

Can't decode payload:[25, 4, 1, 72, 245, 0, 10, 86, 112, 130, 0, 33, 113, 130, 255, 254, 118, 1, 231, 102, 38, 49, 205, 0, 110, 104]

whaleygeek, 2021-07-28:
When I was doing the 2gang testing I added a MockSwitch to help with testing, and I omitted to add the set_switch() method.

Just remove the MockSwitch from your devices.kvs and the problem will go away.

whaleygeek, 2021-07-28:
Note to self - add in set_switch() to MockSwitch (or put in a parent class and inherit)

whaleygeek, 2021-07-28:
Note the "can't decode payload" message is usually because one of your neighbours has a (non energenie) device transmitting on 433MHz. This is informational and should be ignored.

ytoch, 2021-07-28:
David, Many thanks for your help. The Energenie message above came from me. Although the `can't decode payload' message isn't a problem per se, I don't ever see a 'remember device' message as per your instructions, so I am not convinced that my MIHO004 is being properly registered. When I run mihome_energy_monitor.py I see:

Starting energy monitor example
loaded registry from file
REGISTERED DEVICES:
  2gleft -> MIHO009(0x123456,0x1)
  2gright -> MIHO009(0x123456,0x2)
  plus -> MIHO005(0x373)
  dimmer -> MIHO010(0x6c6c6,0x0)
  auto_0x1_0xa56 -> MIHO004(0xa56)
ROUTES:
Logging to file:energenie.csv
<<<TwoBit.SEND
<<<TwoBit.NOTYET
<<<TwoBit.WAIT(2.999960)
<<<TwoBit.SEND
<<<TwoBit.SEND
Traceback (most recent call last):
  File "/home/alarm/local/src/pyenergenie/src/mihome_energy_monitor.py", line 64, in <module>
    energy_monitor_loop()
  File "/home/alarm/local/src/pyenergenie/src/mihome_energy_monitor.py", line 33, in energy_monitor_loop
    d.set_switch(switch_state)
AttributeError: 'MIHO004' object has no attribute 'set_switch'

and in energenie.csv:

timestamp,mfrid,prodid,sensorid,flags,switch,voltage,freq,reactive,real,apparent,current,temperature
1626886242.3694124,4,1,2646,01111000,None,233,50.05078125,-2,34,None,None,None
1626886252.3663929,4,1,2646,01111000,None,233,50.05078125,-2,34,None,None,None
1627370824.786421,4,1,2646,01111000,None,232,49.8515625,-2,35,None,None,None
1627370834.7777684,4,1,2646,01111000,None,231,49.8984375,-2,35,None,None,None
1627370844.7702506,4,1,2646,01111000,None,231,49.8515625,-2,35,None,None,None

Sorry, I appreciate that this issue was originally opened to deal with the MockSwitch issue, so if you want me to open a new issue then let me know. Thanks.

whaleygeek, 2021-07-28:
Hi,

No problem, glad to help. It is possible MIHO004 has suffered the same fate with a missing set_switch method, I will check and fix later tonight for you. It looks like you are receiving data from the device though, so to keep moving you could comment out the d.set_switch in the line reported for now, as I suspect you are more interested in the data.

whaleygeek, 2021-07-28:
P.s. You can alternatively call turn_on and turn_off, as set_switch is just a wrapper that does that when presented with a boolean switch state. I will check and fix properly though.

ytoch, 2021-07-28:
Fantastic, many thanks. I'm not sure whether to be concerned by the 'None' beneath current --the adapter is currently connected to a monitor that is asleep and I'm accessing the Pi remotely at the moment, so I can't plug anything else in to draw a larger current.

whaleygeek, 2021-07-28:
@ytoch The issue was a bit deeper than I had expected.

It turns out that the MIHO004 doesn't have switching capability (monitor only) but was incorrectly specified, so the main loop thought it had a switch, tried to switch it, and of course the method didn't exist.

I have rationalised and corrected a number of issues to do with switch detection, and tested this with a MIHO004, and it logs data to energenie.csv.

This has exposed a different issue to do with accessing data fields from the in memory object when running inside mihome_energy_monitor.py (it doesn't update the in memory object), but it does correctly log data to the .csv file now without crashing.

whaleygeek, 2021-07-28:
```
pi@raspberrypi:~/pyenergenie/src $ more energenie.csv
1627500905.1224356,4,1,482,01111000,None,242,49.8984375,0,0,None,None,None
1627500916.9167469,4,1,482,01111000,None,242,49.8984375,0,0,None,None,None
1627500925.9278092,4,1,482,01111000,None,242,49.8984375,0,0,None,None,None
1627500936.9473183,4,1,482,01111000,None,242,49.8984375,0,0,None,None,None
1627500945.957651,4,1,482,01111000,None,242,49.8984375,0,0,None,None,None
1627500955.0442183,4,1,482,01111000,None,242,49.8984375,0,0,None,None,None
1627501045.995175,4,1,482,01111000,None,243,49.8984375,0,0,None,None,None
1627501435.5457485,4,1,482,01111000,None,241,49.80078125,0,0,None,None,None
1627501444.556968,4,1,482,01111000,None,241,49.80078125,0,0,None,None,None
1627501454.2939837,4,1,482,01111000,None,241,49.80078125,0,0,None,None,None
1627502064.1338756,4,1,482,01111000,None,243,49.8515625,0,0,None,None,None
1627502094.5582776,4,1,482,01111000,None,243,49.8515625,0,0,None,None,None
1627502103.56955,4,1,482,01111000,None,244,49.8984375,0,0,None,None,None
1627502113.1895695,4,1,482,01111000,None,243,49.8984375,0,465,None,None,None
```

In the above log I plugged in a 500W inspection lamp (reports as 465W in the last line).

Note that not all devices report all fields in the CSV - the 5th column has a set of flags indicating which readings are present for the specific device - any flagged as '0' will report 'None' by design.

This is so that the CSV file has a consistent format across all devices.

whaleygeek, 2021-07-28:
@ytoch I think the original set_switch() issue is now fixed based on my testing tonight on my Pi, so I will close this issue now -but see the linked ticket for a somewhat related API issue that I discovered while testing.

closed 2021-07-28 by whaleygeek