whaleygeek/pyenergenie #132: mihome_energy_monitor.py doesn't register or report values on console

whaleygeek, 2021-07-28:
While debugging https://github.com/whaleygeek/pyenergenie/issues/131 I noticed that the mihome_energy_monitor.py example doesn't correctly register the route from a registry retrieved device, to the object created when parsing the registry.

The upshot of this is that all incoming messages are reported as 'no route for unknown device', and therefore if you access the device class instance via the registry iterator, all it's values will be None (as demonstrated by the new print statement below). I never noticed this before in previous testing.

```
pi@raspberrypi:~/pyenergenie/src $ python3 mihome_energy_monitor.py 
Starting energy monitor example
loaded registry from file
REGISTERED DEVICES:
  monitor -> MIHO004(0x1e2)
ROUTES:
Logging to file:energenie.csv
Checking device status
MIHO004(0x1e2) AP:None C:None F:None RP:None RP:None V:None 
Can't decode payload:[37, 87, 172, 14, 161, 185, 81, 142, 114, 176, 156, 17, 63, 16, 172, 63, 235, 224, 189, 38, 87, 35, 217, 187, 147, 5, 145, 137, 26, 98, 247, 134, 59, 3, 38, 97, 210, 217]
Checking device status
MIHO004(0x1e2) AP:None C:None F:None RP:None RP:None V:None 
Checking device status
MIHO004(0x1e2) AP:None C:None F:None RP:None RP:None V:None 
```

whaleygeek, 2021-07-28:
The energenie.csv file stores all received messages from devices that send reports, by virtue of the way it registers a global fsk_router for all messages then appends them to the log.

But really to be useful to users, this demo program probably needs to correctly route incoming messages to the device class instances, so that their RAM variables are correctly updated; besides, the application developer probably wants to use device.get_power() to query the last known power value, etc, rather than just relying on what has been written to the CSV file.

whaleygeek, 2021-07-28:
@ytoch The logging to file works fine now for your MIHO004 (monitor only device), but this new ticket captures an issue that I need to think about and fix, so that you have a useful Python API to query values in a programatic way, rather than always relying on the data dump written to the energenie.csv file.

ytoch, 2021-07-29:
Many thanks David for looking at the problem so quickly. Logging to file is the most important thing for us at the moment but I agree that an alternative would be useful.

whaleygeek, 2021-07-29:
@ytoch No problem, glad to be able to help. Will look into building some better 'get you started' examples for the future, but glad that the log to file functionality gives you what you need for now. Good luck!

tmorley2000, 2022-03-31:
A fairly small change to mihome_energy_monitor.py sets up router for all devices..

https://github.com/tmorley2000/pyenergenie/commit/0c41abe429c1fc7c9225866c10838af861e773ec

whaleygeek, 2022-03-31:
Thanks. I will be picking this project up again in April, feel free to send this as a PR if you want to, and I will merge.