whaleygeek/pyenergenie #133: Is there any way to fix this error on a Raspberry Pi 5?

JamesTimothyMeech, 2023-11-19:
(Virtual_Environment) grappa@raspberrypi:~/Desktop/pyenergenie-master/src/energenie $ python3 radio.py 
Traceback (most recent call last):
  File "/home/grappa/Desktop/pyenergenie-master/src/energenie/radio.py", line 31, in <module>
    libradio                     = ctypes.cdll.LoadLibrary(mydir + "/" + LIBNAME)
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ctypes/__init__.py", line 454, in LoadLibrary
    return self._dlltype(name)
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/ctypes/__init__.py", line 376, in __init__
    self._handle = _dlopen(self._name, mode)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: /home/grappa/Desktop/pyenergenie-master/src/energenie/drv/radio_rpi.so: wrong ELF class: ELFCLASS32

whaleygeek, 2023-11-19:
Yes, run the ./build in that folder and it will rebuild the .so file on your current target machine

JamesTimothyMeech, 2023-11-20:
Thanks for the advice. I got a bit further to this problem:
```
grappa@raspberrypi:~/Desktop/pyenergenie-master/src $ python3 setup_tool.py 
PYENERGENIE setup tool
If it hangs here,
please disable hardware SPI in PREFERENCES/CONFIGURATION

init...
can't open /dev/gpiomem or /dev/mem
```
Does this refer to disabling the SPI in the Raspberry Pi preferences menu? I Have not been able to locate in this repository: PREFERENCES/CONFIGURATION

JamesTimothyMeech, 2023-11-20:
I went ahead and disabled SPI in the Raspberry Pi preferences menu but I get this error:
```
grappa@raspberrypi:~/Desktop/pyenergenie-master/src $ sudo python3 setup_tool.py 
PYENERGENIE setup tool
If it hangs here,
please disable hardware SPI in PREFERENCES/CONFIGURATION

init...
mmap error -1
```
I guess something has in the GPIO changed now they use the RP1 chip for GPIO?

JamesTimothyMeech, 2023-11-20:
I'm wondering if I could adapt this to work on the Raspberry Pi 5: https://github.com/thinking-binaries/pico-energenie

I would also have to adapt it to use the ENER314-RT instead of the Adafruit RFM69HCW but they look like they have the same radio module soldered to them.

whaleygeek, 2023-11-20:
Yes, I suspect the gpiomem has changed completely now that the GPIO is on the south-bridge chip.

The way to fix this would be to look for the latest RPi.GPIO python modules, and dig into the C code for those and see how they access the GPIO, and port that code into the gpio.c inside pyenergenie.

The pico-energenie project was based on this pyenergenie project. The core issue is that in its current incarnation with software spi, you're gonna hit a GPIO mapping issue.

There is a ticket in this repo somewhere where another user ported in the spidev hardware driver and got it working, let me see if I can find that, as that might just work out of the box for you.

whaleygeek, 2023-11-20:
Here is the ticket where another user changed the low level driver to use the spidev kernel driver successfully. 

https://github.com/whaleygeek/pyenergenie/issues/120

If you follow the forks for that user (@Achronite ) or find their fork on their github user account, you might be able to find their code and use that instead. I don't recall a PR coming over for that modification.

The software SPI driver built on top of gpio.c has clearly run out of steam now, so I think your best bet is to change the C code to talk directly to the SPI hardware driver, and that fork mentioned above might be the quickest way to do that.

JamesTimothyMeech, 2023-11-22:
Thanks for the advice. I tried using this fork: https://github.com/Achronite/pyenergenie but I don't think it has the SPI hardware driver code. Because I get the same error. 

I'm now going to trying this one: https://github.com/Achronite/energenie-ener314rt/tree/master ehich claims it can use the SPI hardware driver code although it seems to throw a similar error:
```grappa@raspberrypi:~/Desktop/energenie-ener314rt-master/Examples $ sudo node app.js 
app.js: Initialising board
ener314rt: mmap error -1
app.js: ERROR: Cannot initialise ENER314-RT board, status=-2
app.js: Quitting
```